speeduino/speeduino #998: Staged Injection table mode only considers the first row of the table

theKraid, 2023-02-02:
A friend and I are bench testing a staged injection setup for our naturally aspirated rotaries.
We discovered that in table mode for some reason only the first row of the table is considered for the calculation despite the cursor in TunerStudio moving across the table in height also.

Below is a recording that show the behavior. Pay attention to the injector LEDs as the cursor is moving across the table
Keep in mind this is a TPS-load based (Alpha-N) setup, maybe in MAP-based its fine but we have not tested it this way.
Here's the tune: https://tunes.speeduino.com/#/t/p47vKl3LfI/tune

https://user-images.githubusercontent.com/37898397/216430393-671c8ace-bd09-4208-952d-5ab3670cea40.mp4

noisymime, 2023-02-02:
Have confirmed this, looks like the case for table based staging with TPS as load source was never handled.

I’ll get this added shortly.

noisymime, 2023-02-02:
Fixed now in master and fixes branches.

closed 2023-02-02 by noisymime

theKraid, 2023-02-02:
Awesome! Thank you so much

commit 6f61bd3e (2023-02-02): Allow staged fuel table mode to work with TPS load

Fixes #998
commit c09cb509 (2023-02-02): Allow staged fuel table mode to work with TPS load

Fixes #998