speeduino/speeduino #999: Feature Request: WUE based AFR Target

iltHeo84, 2023-02-03:
During warmup, as the WUE correction tapers down with engine approaching its normal operating temperature, One might like EGO control to engage before WUE reaches 100% (0% correction) to have a "smooth" transition between open loop and closed loop fueling.

Currently EGO control does not take into consideration a richer AFR Target in this situation, so it is constantly fighting against WUE until engine is fully warmed up.

It would be nice to add the option to use **AFRTarget/WUE%** as the target for EGO control.

jonbills, 2023-02-03:
you want ego control to correct to fully warmed up AFRs, before the engine is fully warmed up?why can't you just end WUE at the temp you want it to switch to fully warmed up AFRs?

iltHeo84, 2023-02-03:
> you want ego control to correct to fully warmed up AFRs, before the engine is fully warmed up?why can't you just end WUE at the temp you want it to switch to fully warmed up AFRs?

Quite the opposite, I would like EGO Control to target richer AFRs than warmed up ones

jonbills, 2023-02-03:
Makes more sense :)
I have pid ego control on before end of warmup too, and it just results in it being a bit richer than the target, but stable. does yours oscillate?

iltHeo84, 2023-02-03:
> Makes more sense :) I have pid ego control on before end of warmup too, and it just results in it being a bit richer than the target, but stable. does yours oscillate?

I use Simple with only 2% authority because I mostly use EGO Control to oscillate around stoich when cruising, and help smooth out lean spikes on very little throttle movements that AE cannot handle.
In my case it ends up pegged at the lowest limit, trying to pull AFRs back to warmed-up ones, while with a simple mod it can be made targeting richer AFRs

VitorBoss, 2023-02-03:
2% is effectively locking the correction, the correction resolution is 1% integer

iltHeo84, 2023-02-06:
> 2% is effectively locking the correction, the correction resolution is 1% integer

I understand how authority and different algorithms (Simple or PID) affect EGO correction, but if it is only using warmed-up AFR Targets, it will always be "fighting" againts WUE if enabled before WUE is 100% (zero correction)