speeduino/speeduino #1002: MAP influences pulsewidth when TPS selected as fuelling algorithm

scudderfish, 2023-02-04:
Tune : https://tunes.speeduino.com/#/t/qJ9XsHH84j/tune/settings/engine_constants
Video: https://youtu.be/INBYySeuQ6A

This could be PEBCAK.  The demo is from 202207.1 running with against an Ardustim board.  If I have MAP selected as the fuelling algorithm, changing MAP changes the PW and changing TPS has no effect.  If I have TPS selected as the algo, then changing TPS changes the PW but also changing MAP changes it.  I had a cursory look at the code and I couldn't immediately see anything obvious.

jonbills, 2023-02-04:
have you got multiply ve by map:baro set in your ve table?

scudderfish, 2023-02-04:
You see?  I told you it could be PEBCAK :D

closed 2023-02-04 by scudderfish