speeduino/speeduino #1011: Feature Request: Different sizes of distributor windows as cam trigger

zukhe, 2023-02-24:
There's some models of distributor have different size windows for identifying the first cylinider TDC. If one were able to use the already bult in system, we could get the benefits of sequential injection right out of the box on cars with this kind of system. 

Here's an example of the trigger of my car (it's a 4 cylinder):

1. One window on distributor trigger wheel starts at 72º BTDC and ends right at the TDC (it' has 72º degrees): This is the first cylinder trigger.
2. One window on distributor trigger wheel starts at 72º BTDC and ends 6º BTDC (it has 66º crank degrees); 
3. One window on distributor trigger wheel starts at 72º BTDC and ends 6º BTDC (it has 66º crank degrees); 
4. One window on distributor trigger wheel starts at 72º BTDC and ends 6º BTDC (it has 66º crank degrees); 

So, given that we would need something like this for the "CAM" trigger configuration:

- [ ] A flag for using the same input as both the crank trigger and cam trigger;

Trigger wheel configuration:
- [ ] A "duration of the first cylinder TDC trigger" field and variable;
- [ ] A "duration of the other cylinders trigger" field and variable;

And, of course, a trigger decode for using this info. How feasable is this, guys?

mike501, 2023-02-25:
I'm not sure what your are proposing. Is it,
A) use the distributor as a cam signal alongside a trigger wheel on the
crank to get sequential?
B) use the rising (start) and falling (end) edges of the distributor to act
as teeth for a trigger wheel & measure the size to get TDC and therefore
sequential?
C) something else?

If a) then this already works just use cylinder 4 for the cam pulse.
If b) as the distributor spins at half engine revs you'd end up with a 4
teeth per 360 degrees revolution pattern using a hall sensor and signal
change setting (couldn't use vr). With uneven spaced teeth & measuring the
difference in tooth size the engine speed couldn't change by more than
circa half the time taken to move half the size difference in a revolution
or this risk confusing a tooth for TDC. As it's 6 degrees difference in
total, to give a margin of error we're saying the engine couldn't change by
more than the time it takes to move 3 degrees a revolution. I haven't done
the maths but instinctively this doesn't sound a practical solution to me.
Sorry, Mike

zukhe, 2023-02-25:
It was exactly "b". I see, it would be a tight difference in time between the 1st cylinder TDC window duration and the other ones, maybe unfeasible. I extracted the idea from the maintenance manual of my original ECU, it was a Magnetti Marelli (1AVB). It used the window duration difference as you stated on "b", and worked sequentially with only the distributor as a trigger, some mid 90s kind of thing. That led to me thinking that the delta of the engine speed between triggers in one engine cycle was sufficiently low to work this way.
 
Thanks for your considerations, Mike. I also now think this one is tough to get to work, considering what you've pointed out.

VitorBoss, 2023-02-25:
That is doable but have some caveats, only the injection can be used in that mode and above a certain RPM as the compressions oscillations can cause a false trigger.

zukhe, 2023-02-25:
Guys, for sequential injection purposes wouldn't you need only a reference on start up, once? After cranking you wouldn't need to measure window size, just follow the ignition sequence. Maybe extend cranking time on purpose (without fuel) to get consistent engine speed and a reliable first read.

VitorBoss, 2023-02-25:
Clearly you haven't seen a cranking toothlog of a 60-2 wheel.
The timing is inconsistent on cranking, yeah, the reference need to be done just once, but neet to be done at a consistent timing

TBAMax, 2023-02-27:
As far as I know currently the basic distributor or dual wheel decoder uses only one edge(rising or falling) as a reference. So it is perfectly possible to make a new decoder that uses both edges. This means with the one edge you get it running initially, and the other edge then can be used to sync full sequential. So totally doable. 
  I was messing with the hall sensor distributor at some point maybe it had the same system that all vanes where not equal on purpose, but I did not figure it out, thought that is is just in not accurately made, and discarded it...
  Most likely the consistent edge angle of 72º BTDC that means 90º-72º=18º ATDC can be adjusted earlier so that it is about in the range of 5ºBTDC to 5ºATDC on a another cylinder and then can be used as fixed cranking ignition (ignition is simply fired immediately after vane edge+dwell time). This way initial starting is very easy because it fires right when the vane passes and do not have to wait for any sync or anything. I have seen this used on some VW.

runesmdk, 2023-03-22:
> Clearly you haven't seen a cranking toothlog of a 60-2 wheel. The timing is inconsistent on cranking, yeah, the reference need to be done just once, but neet to be done at a consistent timing

Forgot to tag you in previous reply

runesmdk, 2023-03-22:
Using this, you could even go as far as using the edge with the same distance as starting trigger angle. That would take that part out of the equation? I think the daihatsu+1 locks it to using the edge as well so theres no adjusting starting ignition angle.

runesmdk, 2023-03-22:
And that actually leans a bit up to #953 as well

runesmdk, 2023-03-22:
@VitorBoss 
I've got a hall distributor as well with a 4-windowed shutterwheel. Either making the windows to match daihatsu 4+1 or like the original poster here suggested slimming one window down and looking at both edges (This could almost double the trigger resolution, like 8-1 if the slim one could get singled out?) would be huge improvement, and would enable me to just use the distributor as a triggersensor alone, and go sequential both ignition and injection.
Using this, you could even go as far as using the edge with the same distance as starting trigger angle. That would take that part out of the equation? I think the daihatsu+1 locks it to using the edge as well so theres no adjusting starting ignition angle.
And that actually leans a bit up to https://github.com/noisymime/speeduino/issues/953 as well.

VitorBoss, 2023-03-22:
For English use tradutor from portuguese

Cara, eu entendo como funciona, tem limitações para o que pode ser feito, mas não é impossível, porém alguém que não tem esse motor desenvolver esse código é extremamente difícil, a maioria foi adicionado por pessoas que usam o decoder, se você sabe fazer ou conhece alquem que faça com certeza é bem vindo, se não tem o https://app.bountysource.com/teams/speeduino onde você oferece uma recompensa para alguém fazer por você.

runesmdk, 2023-03-23:
@VitorBoss What about converting the daihatsu+1 which is already present to enable sequential on that? Right now its flagged as not sequential.

runesmdk, 2023-03-23:
> @VitorBoss What about converting the daihatsu+1 which is already present to enable sequential on that? Right now its flagged as not sequential.

Or even going as far as how megasquirt 1 did it, by simply defining every triggeredge's angle directly in tunerstudio?