speeduino/speeduino #1013: Feature Request: Using Serial1 as the primary output

RempageR1, 2023-03-01:
Hi,

I`m running a Teensy3.5 which doesn't support Serial out of the USB and the first serial port available. So for USB, you need Serial and for Bluetooth/Wifi you need Serial1. 

I`ve got the Serial1 working with TunerStudio, but currently that does require a lot of find/replaces in the code to change all the Serial references to Serial1. 

Would it be an option to add a definition like this:
#define serialPort Serial

And then change all the Serial.x commands to serialPort.x.

That way, users can replace that single line if they want to change the primary serial interface.

I`d be happy to provide the pull request for this, but before I do I would like to see if it might make it into the mail code or not. 

Thank you.

jankovalski, 2023-03-01:
You can just put #define Serial Serial3 in globals.h and that's enough. Just remember to use Serial3 or higher because serial 1 and 2 has that FIFO thing which makes it incompatibile with speeduino code.

RempageR1, 2023-03-02:
Hi jankovalski,

Thank you for your help with this one, and indeed the #define seems to do the trick for the most part. Since I have the Teensy adapter board for the 0.4.4c and it only has serial1 ports, I want to use Serial1.

And I actually got Serial1 working by using the #define command in globals.h and adding 'Serial.begin(115200);' to line 300 of the init code:
https://github.com/noisymime/speeduino/blob/master/speeduino/init.ino#L300

It has to be just below the pinmapping config for some reason. If you do that, then Serial1 works just fine. 

I`ll close this one as the renaming of Serial.x is not required.

closed 2023-03-02 by RempageR1

giovannni97, 2024-01-04:
> You can just put #define Serial Serial3 in globals.h and that's enough. Just remember to use Serial3 or higher because serial 1 and 2 has that FIFO thing which makes it incompatibile with speeduino code.

Hi, I have a similar issue trying to use Serial1 for a GPS module, could you explain what "FIFO thing" does Serial1 present?
Because i seem to get serial data but some byte are missed and this might be why.