speeduino/speeduino #1015: mistake on error.h ?

supernono-fr, 2023-03-04:
hello 

i think there is a mistake on file  error.h on line 24 

#define ERR_DEFAULT_CKT_SHORT   80 //Note that the default is 40C. 80 is used due to the -40 offset

i think CKT is wrong, CLT must be right

noisymime, 2023-03-14:
Yes, this is correct however `ERR_DEFAULT_CKT_SHORT` (along with a number of other defines in this file) are not actually used anywhere. Ideally this would get fixed in the redesign of the whole error system, but I haven't made much of a start on this yet. 

I'll fix the typo on this for now, but no impact from the error at all.

supernono-fr, 2023-03-14:
OK thank you for your answer, realy interested by the error system can i do something to help ?

adbancroft, 2026-09-06:
Closing - non-issue that's out of dare

closed 2026-09-06 by adbancroft