speeduino/speeduino #1021: Ignition Output Appears to stay low between events

Joraeim, 2023-03-14:
I have a B18C1 w/ a stock distributor. The output signal from the ECU (DropBear) pulses high at an ignition event. This appears to be the incorrect direction even when I select "Going Low" in the ignition settings.
[image]

Per the reference I found below and my own experience, the output should be high and pulse low.
[image]

I'm still investigating and this may be a setting I am missing, but as may be expected, this behavior fried the igniter.

noisymime, 2023-03-14:
"Going Low" refers to the coil is designed to fire when the signal from the ECU goes to ground. When the signal goes high the coil will begin charging (ie dwelling) and it will fire when the signal drops back to low. This is 95+% of modern ignition systems, but it looks like the one you're describing is the opposite. 

If you need the coil to fire when the signal goes high, then you need to select the 'Going High' option. This will invert the standard behaviour and output polarity (ie the signal will be normally high, then go low to begin the dwell period and then go high again to fire the coil)

Joraeim, 2023-03-14:
@noisymime, 

Thanks for the quick response. 

On my system, the ignition output needs to remain high except for the short firing pulse (firing time).  I tested both the "Going High" and "Going Low" settings and only got what appeared to be a "Going High" pulse on the oscilloscope.

I believe I may have found the bug.

ScheduledIO.h defines the coil#chargingDIRECT() using a ternary operator to take the IgInv setting into account.
[image]

However, the DropBear HW appears to use the MC33810 IO expander which does not use the IgInv setting.
[image]

I'll make the modification, recompile, test, and get back to you.

noisymime, 2023-03-14:
Ahhh good pickup, I have no idea how or why that got missed on the MC33810 code! If the addition of that check resolves your issue please let me know and I'll get this added ASAP

Joraeim, 2023-03-15:
Adding the check did resolve my issue. I've attached a screenshot of the oscilloscope output below. I only verified one output.

[image]

Here are the lines I updated in "scheduledIO.h". For some reason, the Arduino compiler didn't like the ternary operator on these lines. I don't know why.

`// Define Coil charging and discharge polarity based on Spark Output Trigger direction (configPage4.IgInv).
// For some reason, the Arduino compiler doesn't like a ternary operator here.
#define coil1Charging_MC33810()      if(configPage4.IgInv == GOING_HIGH) {coil1High_MC33810();} else { coil1Low_MC33810();}
#define coil1StopCharging_MC33810()  if(configPage4.IgInv == GOING_HIGH) {coil1Low_MC33810();}  else { coil1High_MC33810();}
#define coil2Charging_MC33810()      if(configPage4.IgInv == GOING_HIGH) {coil2High_MC33810();} else { coil2Low_MC33810();}
#define coil2StopCharging_MC33810()  if(configPage4.IgInv == GOING_HIGH) {coil2Low_MC33810();}  else { coil2High_MC33810();}
#define coil3Charging_MC33810()      if(configPage4.IgInv == GOING_HIGH) {coil3High_MC33810();} else { coil3Low_MC33810();}
#define coil3StopCharging_MC33810()  if(configPage4.IgInv == GOING_HIGH) {coil3Low_MC33810();}  else { coil3High_MC33810();}
#define coil4Charging_MC33810()      if(configPage4.IgInv == GOING_HIGH) {coil4High_MC33810();} else { coil4Low_MC33810();}
#define coil4StopCharging_MC33810()  if(configPage4.IgInv == GOING_HIGH) {coil4Low_MC33810();}  else { coil4High_MC33810();}
#define coil5Charging_MC33810()      if(configPage4.IgInv == GOING_HIGH) {coil5High_MC33810();} else { coil5Low_MC33810();}
#define coil5StopCharging_MC33810()  if(configPage4.IgInv == GOING_HIGH) {coil5Low_MC33810();}  else { coil5High_MC33810();}
#define coil6Charging_MC33810()      if(configPage4.IgInv == GOING_HIGH) {coil6High_MC33810();} else { coil6Low_MC33810();}
#define coil6StopCharging_MC33810()  if(configPage4.IgInv == GOING_HIGH) {coil6Low_MC33810();}  else { coil6High_MC33810();}
#define coil7Charging_MC33810()      if(configPage4.IgInv == GOING_HIGH) {coil7High_MC33810();} else { coil7Low_MC33810();}
#define coil7StopCharging_MC33810()  if(configPage4.IgInv == GOING_HIGH) {coil7Low_MC33810();}  else { coil7High_MC33810();}
#define coil8Charging_MC33810()      if(configPage4.IgInv == GOING_HIGH) {coil8High_MC33810();} else { coil8Low_MC33810();}
#define coil8StopCharging_MC33810()  if(configPage4.IgInv == GOING_HIGH) {coil8Low_MC33810();}  else { coil8High_MC33810();}`

noisymime, 2023-03-15:
Ahhh that rings a bell, I suspect the ternary issue was why this was initially left out for the MC33810, but then never re-evaluated. 

I'll play with a couple of solutions and commit the preferred one shortly. Thanks for the confirmation and testing.

noisymime, 2023-03-15:
Just to check, which way did you have the Spark Output trigger setting in your testing? 

The code you've posted above is the opposite of what I would expect to see. To align with the existing code and the intent of that setting I would expect this to be (For Eg):
```
#define coil1Charging_MC33810()      if(configPage4.IgInv == GOING_HIGH) {coil1Low_MC33810();} else { coil1High_MC33810();}
#define coil1StopCharging_MC33810()  if(configPage4.IgInv == GOING_HIGH) {coil1High_MC33810();}  else { coil1Low_MC33810();}

Joraeim, 2023-03-15:
You are correct. I currently have it set to "Going Low".

noisymime, 2023-03-16:
Fix committed for this now to `master` and `-fixes` branches. Thanks for the testing! 

Just for reference, your setup should be on the 'Going High' setting now.

commit 1886eb38 (2023-03-16): Honour the Inverted ignition output setting on boards using MC33810. Fixes #1021
closed 2023-03-16 by noisymime