speeduino/speeduino #1029: Spark Table using MAP KPA for load when TPS Control Algorithm is selected

MykkH, 2023-04-07:
Spark Table using MAP KPA for load when TPS Control Algorithm is selected. 

At first I thought it just wasn't changing the Y-axis labeling, however it still very much used the MAP load values with engine running.

jonbills, 2023-04-07:
Do you have TPS set as load index in spark settings? 
Add your tune and a log to the issue.

MykkH, 2023-04-07:
Settings >Engine Constants >Control Algorithm: TPS, is that not the blanket way to set up Alpha-N?

LPG2CV, 2023-04-07:
VE and spark loads are independant of each other. Look in Spark settings :)

noisymime, 2023-04-07:
That's working as intended. The option in Engine Settings is for the fuel algorithm with spark having its own load source setting

closed 2023-04-07 by noisymime