speeduino/speeduino #1030: Staged injection and injector duty limits

scudderfish, 2023-04-09:
I don't know if this is an issue, or just "that's the way it is".  I have a staged injection set up and an injector duty limit of 80%.  When I hit 80% the secondary injectors kick in.  However I can end up overcranking the secondaries with no limit.  On the bench my ECU is currently running the primaries at 80% and the secondaries at 95%

Should the secondaries respect the duty limit as well?

VitorBoss, 2023-04-09:
Currently the secondary injection aren't limited at all. Usually the secondary injection have bigger injectors, that is why no limit is in place.

jonbills, 2023-04-10:
there's no pwlimit applied even to the primary injectors in table mode.
It would be pretty easy to apply pwlimit to both primary and secondary in both modes.

scudderfish, 2023-04-10:
OK, I'll have a dig in the code and create a MR

VitorBoss, 2023-04-10:
You can't just apply a limit for the injectors duty, if you do, you will have a lean mixture.

The correct path is to enable fuel/ignition cut (like Rev limit) when above the duty limit to protect the engine.

jonbills, 2023-04-11:
The current logic for non-staged fueling does just limit the DC, I'm pretty sure. Maybe exceeding the injector duty should just be a warning somewhere.