speeduino/speeduino #1043: if sparc setting is set to ROTARY going high will not be programmed to the controller

transi01, 2023-05-01:
If sparc setting is set to Rotary switching the the output triggers to: going high doesn't work!

If the trigger will be set to going high and config will be burned it switches itself back to going low after powering down, which is fatal for for example OEM coils and ignition module of Mazda RX7 FD's.

I have reproduced the problem using the following firmware versions on different Mega2560's: 
Have tried it with Rotary settings set to FD and RX8, still the same :-(

202202, 202207.1 and actual master (all flashed using SpeedyLoader-x64 1.5.1)

Erasing flash didn't help, have also used a brand new Mega2560 for testing with the same result.

If I set sparc setting to "Wasted Spark" the switching from going low to going high works as it should!

Thanx in advance,

// Thomas

noisymime, 2023-05-01:
Both the FD and RX8 OEM ignitions should use `Going Low`. They expect a High signal for the duration of the dwell period and then will spark when the signal is brought back to ground. I’ve never seen any stock FD or RX8 system that would require `Going High` to operate. 

How exactly do you have your coils wired up?

transi01, 2023-05-03:
Well, I've designed a PNP ECU for the 16BIT RX7 FD3S and I'm just on starting for tests.

 

I've measured the signal input of the OEM Ignition module and when just ignition switch is on it is LO, so for my understanding the setting for the output must be "going high"?!

When I measure the IGN module input on a 1993 1.6 MX5 NA it is high, when only ignition is switched on and here it must be set to going low!

 

But anyway shouldn't it be possible to switch it over from "going low" to "going high" as in other configurations also?

 

// Thomas

 
 

Gesendet: Montag, 01. Mai 2023 um 16:02 Uhr

noisymime, 2023-05-11:
Where are you measuring, the input or the output of the ignition module?

The RX7 and MX5 ignitions work on identical polarity. Namely the ECU signal will idle low and only go high to begin the dwell phase. If you're measuring High on a NA MX5 with just the ignition on (Engine not running) then either the signal is wrong or more likely, you're measuring the incorrect signal.

noisymime, 2023-06-02:
Closing this as Going Low is confirmed as the correct setting for these rotary setups.

closed 2023-06-02 by noisymime