speeduino/speeduino #1045: Feature request: Open idle valve at full throttle

zukhe, 2023-05-08:
Create a function that adds an option for opening the idle valve/stepper motor when 100% TPS is achieved, allowing an additional air passage to feed the engine.

digmorepaka, 2023-05-09:
What's the benefit/goal here?

Nocnafuria22, 2023-05-10:
> What's the benefit/goal here?

More air for N/A engines. In my case when i rev engine over 5k rpm i can see map spikes from 100kpa to +/- 95kpa at full throttle.

digmorepaka, 2023-05-10:
> > What's the benefit/goal here?
>
> More air for N/A engines. In my case when i rev engine over 5k rpm i can see map spikes from 100kpa to +/- 95kpa at full throttle.

Have you tried measuring the pressure in front of the throttle? Often this is caused by the filter/air box. I'd also be interested to see your results if you did actually open the idle valve - if it helps with your setup in any capacity. If it's a regular PWM valve and you don't want to mess around in software, you can hook up a mosfet in parallel (such as one of the high current outputs) with the idle fet and setup a programable output to turn it on when WOT.

Nocnafuria22, 2023-05-10:
> If it's a regular PWM valve and you don't want to mess around in software, you can hook up a mosfet in parallel (such as one of the high current outputs) with the idle fet and setup a programable output to turn it on when WOT.

I already tried this but with a 3 wire IACV i am not able to fully open the valve. I would have to use a relay to cut off the second signal to IACV. I will try measuring the pressure in front of the throttle. It is currently measured at a distance of 20 cm from the throttle.

closed 2026-09-06 by adbancroft