speeduino/speeduino #1050: Feature request: Flex based starting settings

ElroyDroid, 2023-05-23:
Would be great to see the priming pulsewidth, cranking and after start enrichment setting adjustments able to be set by flex fuel percentage.

For decent cold starts on E85 I have to max out the prime pulse at 120ms and 1000% cranking enrichment to get the engine fired up close to 0deg C, but on regular 0% ethanol "unleaded"  I only need say 200% enrichment and 5ms prime pulse at the same temperature for example. At the moment I'm just using the "Load dialog settings" and loading different settings when switching between E85 and unleaded.

A priming pulsewidth modifier and cranking enrichment modifier by flex percentage would be helpful.

wilmardo, 2023-05-23:
Would #612 cover your usecase? Your case sounds somewhat similair to @VitorBoss his description :)

ElroyDroid, 2023-05-24:
That will help. I am new to this and I can see the changes VitorBoss made, I can see where it was committed, but I've cloned the latest code, verified it in Arduino software and flashed it, updated the ini file in TS but I cant see those options for PW adder.

Do I have to manually add VitorBoss's code to gain PW adder and extra spark event on cranking? I'm new 😂

wilmardo, 2023-05-30:
You should clone this repository:
https://github.com/VitorBoss/speeduino/tree/enhaced-cranking

And then switch to the `enhaced-cranking` branch (`git checkout enhaced-cranking` or somewhere in your git UI)

That should have the updated code which you should compile and flash. Then load the updated ini in TS.

ElroyDroid, 2023-05-31:
Thankyou, this helped massively. I figured it out.

But it looks like this will only enable a toggle for a priming pulse multiplier to gain more than the 120ms maximum prime pulse, does not appear to scale by ethanol content that I can see. I'll give it a try with the extra spark, but I would like to somehow scale the prime pulse by ethanol content combined with engine temperature. I suppose this is an easy way to just enable or disable the multiplier when going from unleaded to E85

wilmardo, 2023-05-31:
Nice, glad you got it sorted! 
If you could report your findings with the updated code in the PR (https://github.com/noisymime/speeduino/pull/612), that would help making it better and get it to a release.

> does not appear to scale by ethanol content that I can see

This could still be a valid feature request, I just thought the PR might help :)