speeduino/speeduino #1051: HalfsyncToFullSync issues found on testing.

TBAMax, 2023-05-24:
When switching from half to full sync at engine startup, the closeInjector and openInjector assignments are changed. But this creates the possibility that when this is done during the fuel or ignition pulse, the started pulse on some channel is not turned off. This shows up occassionally on logic analyser when testing(image below). 
Indicated by very long injection pulse. This happens only occassionally, but will probably flood the corresponding cylinder.
Probably this might be  already a known issue.

[image]

VitorBoss, 2023-05-24:
Yes, that could happen if the CAM sensor came after a pulse started, this is easily avoided by checking for stopped timers before changing.

TBAMax, 2023-05-25:
> 

> Yes, that could happen if the CAM sensor came after a pulse started, this is easily avoided by checking for stopped timers before changing.

Ok, I am on it. Hoping to get the PR submitted soon.

noisymime, 2023-05-26:
Fixed with #1052

closed 2023-05-26 by noisymime