speeduino/speeduino #1053: WMI and VVT pin assignment confilct

tuxit09, 2023-05-26:
First as you will see in the youtube video, whatever I change pin assignment for VVT then WMI will change automatically. In the opposite direction as well. I found that this also takes place if working in offline mode or connected I presume this is a mapping issue in TS or is this a speeduino configuration problem??

Please see:
https://youtu.be/zorqhtIsj_s

Tunerstudio Version:  TS MS Ultra v3.1.08
Speeduino Board: UA4C White Fang WTMtronix
Speeduino Firmware: 202207.1

tuxit09, 2023-05-26:
Is this in any way related to issue #582? Only correlation for me is I'm using pin 5 for WMI. Thanks again.

VitorBoss, 2023-05-26:
Currently you can only use one at time, both uses the same code internally, I made a PR(#845) to solve this, hope @noisymime analyse it for next release.

noisymime, 2023-05-26:
As mentioned, WMI and VVT cannot function together in the current firmware. I've just pushed up a warning for this in the ini, but the ultimate solution here is #845. 

I'll keep this open as I expect to merge that PR early in the next dev cycle

tuxit09, 2023-05-28:
> As mentioned, WMI and VVT cannot function together in the current firmware. I've just pushed up a warning for this in the ini, but the ultimate solution here is #845.
> 
> I'll keep this open as I expect to merge that PR early in the next dev cycle

Did this change in the .ini file limit the user from configuring VVT and WMI simultaneously? 

VitorBoss made a modification to test moving WMI to VVT2 function opening the potential for VVT1+WMI but I think this was prevented from testing by the sanity check found in the newest release for VVT+WMI conflict. 

I opened the .ini but was not sure how to modify with breaking something along the way.

noisymime, 2023-06-02:
> Did this change in the .ini file limit the user from configuring VVT and WMI simultaneously? 

Yes it did, and this is intentional given the current timer arrangement.

> VitorBoss made a modification to test moving WMI to VVT2 function opening the potential for VVT1+WMI but I think this was prevented from testing by the sanity check found in the newest release for VVT+WMI conflict. 

When that PR is brought in I'll remove the newly added check. It wasn't going to make it in for the last release though, so better to have the warning at least

tuxit09, 2023-07-03:
I just wanted to update this issue/topic in hope that VictorBoss's patch makes it to main stream for enabling VVT1 and WMI simultaneously. This seems to be working just perfect for me with the exception of a table values isn't persistent and must send tunerstudio settings to ecu each connect. Otherwise, works.

tuxit09, 2023-11-20:
Just wanted to bump this topic/issue. I have been running the patch for this that VitorBoss worked up. It works flawless except the WMI table data is corrupted on reboot of the ecu or connect of tunerstudio I'm not sure. For whatever reason the values just become random. Having tested this for months on reliability I can confirm with this patch Single VVT with WMI enabled works just great otherwise.

noisymime, 2024-03-15:
Closing as this is now resolved with #845