speeduino/speeduino #1054: Flood clear operation not reliable.

TBAMax, 2023-05-27:
It is almost impossible to perform flood clear, because when there are ignitions from residual fuel the engine will briefly start on one or two cylinders and go out of cranking mode. This causes the flood clear to cancel, and engine keeps running on one or two not flooded cylinders and remaining already flooded cylinders are flooded even more (and engine oil also quickly diluted with fuel).

The proposition is to make changes to the flood clear so that it keeps being engaged until the throttle is released to below flood clear value (currently flood clear mode cancels immediately when rpm raises above cranking).

TBAMax, 2023-05-27:
Also there seems to be a spike in the RPM before the half sync or full sync is achieved with the missing tooth decoder, this means there is some additional fuel already injected during that, so this also would need to have the flood clear to be latched on. 
I do not know if others also experience the rpm spiking before sync? I was just looking at the gauge in the TS. And the flood clear seemed to flicker off and on there too. Might be the tacho sweep on the TS gauge, not sure.

jonbills, 2023-05-27:
i think it was discussed on discord some time ago, the priming pulse puts fuel in the cylinders regardless of flood clear.
Maybe the change should also not do priming pulse when flood clear set.

jonbills, 2023-05-27:
Looks like I'm wrong, injector priming respects flood clear in master at least.

noisymime, 2023-06-02:
Would it be sufficient to simply turn the ignition off as well as fuel? That would prevent any premature starting as the fuel clears.

I've never run into the issue you describe with it, but I can see how it may be a problem.

gdiciocco, 2023-07-11:
I've almost blown an engine relying on floodclear. 

One tends to use it to load oil in engines with dry sump, and having the engine start-up and overrev without oil is something which makes you invoke ancient and current divinities. 

I went saying it in discord months ago but looked like I were the only lucky one. 

Better late than never.

TBAMax, 2023-07-31:
For me it is logical to leave the ignition running to burn off any exess fuel. But just have some laching mechanism in place that the flood clear do not come off until accelerator pedal is releaced back below the flood clear level. Currently flood clear cancels when RPM reaches some treshold. It would be preferable that flood clear only cancels when the throttle is reduced.

TBAMax, 2023-07-31:
Another idea is to leave ignition running AND retard the ignition angle. The retarded ignition angle would help to clear the cylinders more easily -- easier to ignite the left fuel (and also less stress to the engine in case of actual ignitions).