speeduino/speeduino #1056: PWM idle with 3 wire IACV not working after update

Nocnafuria22, 2023-05-31:
After update to 202305 or 202207.2 PWM 3 wire idle valve stop working. When back to 202207.1, it works again. I noticed that it work up to 202210 master release from 07.04.2023. In next master release it stoped working.

noisymime, 2023-05-31:
Can you share the tune that you're using? There was nothing really committed around that time that would impact idle control and definitely nothing that was backported to 202207.2. 

It could be related to a pin mapping though, so just want to check your tune to see how you've got it configured.

What exactly stops working? Is it both the outputs, only one of them etc?

Nocnafuria22, 2023-06-01:
> It could be related to a pin mapping

You was right, I totally forgot about change it after update. Thanks for advice.

closed 2023-06-01 by Nocnafuria22