speeduino/speeduino #1057: Feature request: semi sequential/waste spark on cranking only

ElroyDroid, 2023-06-01:
Not sure if it's possible, but I have an expensive ECU on my other vehicle which has a "quick start" mode. Even when sequential injection and direct fire ignition is set up, when the engine is below the cranking RPM you have set it will fire injectors in semi sequential mode and ignition in waste spark mode to assist in faster starting. 

I did notice before I had my cam signal connected running in wasted spark and semi sequential mode the engine would start quicker.

Not necessary but a  cool concept.

ElroyDroid, 2023-06-02:
Ahh ok didn't realize

I'm using Toyota 36-2 with 4-1 cam trigger

noisymime, 2023-06-02:
Yep, this will already be active then. Look in TunerStudio for the indicators called 'Sync' and 'Half sync', they will show you what state it's currently in (Including in the log). 

When 'Half Sync' is active, you'll get semi-seq and wasted spark.

closed 2023-06-02 by noisymime