speeduino/speeduino #1058: Add VSS to launch

runesmdk, 2023-06-09:
Add VSS to launch;

Add a setpoint for which, if all conditions are met, launch is active below set vss. 
Make it so it is only enabled with a tick-box, so people not running a vss won't get false positives.

bossthetuga, 2024-01-10:
Great feature to be added!
I made it work by programmable outputs.

If VSS lower than 5, pin 51 goes LOW.
On launch control setting, activate it by pin 51 going LOW and use FLOAT.

runesmdk, 2024-01-11:
> Great feature to be added! I made it work by programmable outputs.
> 
> If VSS lower than 5, pin 51 goes LOW. On launch control setting, activate it by pin 51 going LOW and use FLOAT.

Thats actually very clever 👌👌👌

ElroyDroid, 2024-01-11:
> Great feature to be added! I made it work by programmable outputs.
> 
> If VSS lower than 5, pin 51 goes LOW. On launch control setting, activate it by pin 51 going LOW and use FLOAT.

Genius! I've wanted this for a while, this never occurred to me. Thank you 🙏

bossthetuga, 2024-01-11:
Thank you guys!!!

If you guys like to watch videos, here is my Youtube channel: https://www.youtube.com/c/BossGarage
I make videos about speeduino (and more) and showed that some months ago, just don't forget to enable subtitles.

Have a great day ✌

runesmdk, 2024-01-11:
> Thank you guys!!!
> 
> If you guys like to watch videos, here is my Youtube channel: https://www.youtube.com/c/BossGarage I make videos about speeduino (and more) and showed that some months ago, just don't forget to enable subtitles.
> 
> Have a great day ✌

There's many videos to look through. Can you specify which one?

bossthetuga, 2024-01-11:
> There's many videos to look through. Can you specify which one?

Speeduino on Nissan 200sx: https://www.youtube.com/playlist?list=PL45KNTE54f_wvr_l-5jG52W1o28P97YAj
Speeduino on dirt bike 125cc: https://www.youtube.com/playlist?list=PL45KNTE54f_xnzlSSqZHpfCisOr2G_bFO
Speeduino on Honda engines: https://www.youtube.com/playlist?list=PL45KNTE54f_zQLX6nNLQk8BbUJtGLWFaY
Speeeduino digital dashboard: https://www.youtube.com/playlist?list=PL45KNTE54f_xjOh1gHrcI6Z11KJu1PwVA

The launch control by VSS on speeduino is: https://www.youtube.com/watch?v=HR3slVr7Qqw&list=PL45KNTE54f_zQLX6nNLQk8BbUJtGLWFaY&index=6

runesmdk, 2024-01-11:
> Speeduino on Nissan 200sx: https://www.youtube.com/playlist?list=PL45KNTE54f_wvr_l-5jG52W1o28P97YAj Speeduino on dirt bike 125cc: [...]

In this video, you left it on pullup instead of floating. Did you change that later? Also, it seems your vss input is all over the place when you start to roll back, jumping up to nearly 180kmh

bossthetuga, 2024-01-11:
> In this video, you left it on pullup instead of floating. Did you change that later? Also, it seems your vss input is all over the place when you start to roll back, jumping up to nearly 180kmh

When configuring the programmable output, I miss edited the VSS pulses.
But yes, it has a little noise, but it works.

I changed it after to FLOAT since it's a programmable output.
Pull-up is for a ground wire switch, pull-up stays always 5v when not switched on.

ElroyDroid, 2024-01-20:
I have my clutch switch setup with a resistor pulling up to 5V and it grounds when clutch is pushed in.

The only way I could make launch work when I set it up was have launch set to clutch  switch active high and pullup enabled in TS. It seems to work backwards from the Speeduino instructions of clutch enabled when low and float. 

It works perfectly this way for launch and flat shift though.

I set up the programmable output as active high when VSS < 5 and also added a second condition "OR" RPM > 5000rpm so my flat shift would still work. When I only added the VSS < 5 I didn't think about the flat shift using the same pin so essentially did a clutch kick and lost traction changing gears bouncing of the rev limiter haha.

My launch and flat seems to work only with the settings in Tunerstudio set opposite to you guys.

Just remember to add an RPM condition if you're also using flat shift .