speeduino/speeduino #1064: Autotuning secondary fuel table

ninjawarrior821, 2023-06-27:
My turkey brain thought it would be a good idea to use the secondary fuel table for the boost region of my tune. I set up the secondary table to 'Switched - conditional' above ~100 kPa and tried to use Tune Analyze Live on the boost map (imported). Unfortunately, the car just solely used the boost map to run, therefore idling way too rich, and I gave up there.

Maybe I just set it up wrong, maybe it's a problem with Tuner Studio and not Speediuno software, just thought I'd ask.

ElroyDroid, 2023-06-29:
You aren't able to use Tune Analyze Live on the secondary fuel table. When you imported your boost map it would have overwritten the main fuel table with this.

You're better off using VE analyze in MegaLogViewer and selecting vetable2, but you'll need to log and analyze, you won't be able to do it live.

Is there any reason you're using secondary fuel table just for boost? You can get plenty of resolution for most applications using one fuel table.

ninjawarrior821, 2023-06-29:
Do you know if not being able to use Tune Analyze Live on the secondary table could be changed with 3rd party Speeduino profile, or if it's something I'd have to go to Tuner Studio to ask about?

I forgot about VE analyze in MegaLogViewer for this application.

No reason, thought it'd just be fun to try and the Conditional-Switched functionality seemed like a route to do it. I'm currently tuning with just one table now.

ElroyDroid, 2023-06-29:
I'm fairly new to this, as far as I can tell it is Speeduino limiting the live tuning to one table. I've seen screenshots of Tunerstudio running Megasquirt with multiple "Fuel table 1", "Fuel table 2", "Fuel table 3", etc as tabs under Tune Analyze Live.

bossthetuga, 2024-01-10:
Auto tune for the secondary fuel table would be great.
My idea is the same as yours, to use it for the boost region!

hani93, 2025-01-02:
It is possibe and i did it about year ago. i will look and guide you in few days, what needs to be modified to get second table shown up in autotune view

LPG2CV, 2026-09-07:
Could the higher kpa table become the primary table having moved the tuned lower kpa table to the 2nd table. Then use logic to switch between tables. Then able to use autotune to tune higher kpa in the primary table.