speeduino/speeduino #1078: RPM2 as VSS not working with cam speed trigger

zuripig, 2023-07-27:
Using Pin 18 as VSS input does not seem to work, Hall VSS connected to RPM2, with only a camshaft mounted missing tooth pattern.

Trigger Pattern - Missing Tooth
Primary Base Teeth - 36
Primary Trigger Speed - Cam Speed
Missing Teeth - 1

I'm using the latest master branch firmware on a mega based board, 4 cyl sequential fuel and ignition.

Using Pin 2 was working with a modified ini file.

noisymime, 2023-07-28:
Testing OK here. What board are you using (and hence what pin for RPM2 / VSS)?

zuripig, 2023-08-01:
I've had it apart again, double checked everything hardware wise. Jumpers for RPM2 are correct for a hall sensor. Works fine on pin 2 and on pin 20. Pin 18 does not work for me. 
Wired up VSS with a friend over the weekend on his 0.4.3c (VR sensor through a conditioner) and he also does not see anything for VSS when using pin 18. He is using missing tooth crank speed only.
Here's my tune attached
Btw, running latest master again (ee807bb) and flat shift/launch now both work with the rolling limiter :)

[2023-07-31_22.33.06.msq.zip](https://github.com/noisymime/speeduino/files/12234067/2023-07-31_22.33.06.msq.zip)