speeduino/speeduino #1086: Idle integral reset rpm hysteresis not working causing stall

155ac, 2023-09-02:
The integral reset rpm hysteresis option is not working, making the engine stall in deceleration.

The iac value should be held until rpm drops below the rpm hysteresis setting then the pid should start.

[image]
Here the iac value should come from the iac stepper motor until the rpms drop below 1110 rpm, then the pid should start.

Here are my idle configs.
[image]

VitorBoss, 2023-09-02:
That is very odd, the check is in place and is right AFAIK
https://github.com/noisymime/speeduino/blob/master/speeduino/idle.ino#L719
Also, don't use 0 as minimum steps as it will definitely stall the engine, try something like 10.

155ac, 2023-09-02:
the minimum iac value is near zero because i adjust the throttle stop screw in a way that wen the engine is warm and with no load the value stay close to zero.

the problem is when is load on the engine i stalls because the pid is closing the valve very fast, because ther is no rpm hysteresis. When it should use the table and then it slowly going back to pid control, like how he does when the tps is above the reset. as you can see in the log as i close the throttle it go slowly from table to pid control, but is not doing based on rpm as it should.

Only the tps value is working, i try to increase and decrease the rpm hysteresis in settings, but it's still not working.
[image]

155ac, 2023-09-02:
I'm also running the latest build 2023.05 and the idle work great in open loop, only in closed loop the problem appears so no hardware problems in the stepper motor and driver.

paoloponzano, 2024-06-07:
> the minimum iac value is near zero because i adjust the throttle stop screw in a way that wen the engine is warm and with no load the value stay close to zero.
>
>
>
> the problem is when is load on the [...]

I have the same issue, was a fix for this found?