speeduino/speeduino #1104: O2 feedback based on load

Admanirv123, 2023-09-27:
Allow o2 feedback based on load and look up of afr table as opposed to just current tps. Will aid spool by running leaner mixture, but suggest 6 or 7 psi max for throughput/cpu limitations of the 2560 in altering mixture in timely manor.

VitorBoss, 2023-09-27:
Are you using alpha-N fuelling?

Admanirv123, 2023-09-27:
No I am not.

Admanirv123, 2023-09-27:
Hi, for example if I had requested 12.6 afr all the way to 150 kpa in afr target I have to be x throttle defined in afr/o2 settings typically 100%. But I don't want to run 12.6 afr all the way in max boost ie target 250kpa in full throttle as its too lean, hence why a switch off would be good ie max o2 closed loop feedback (active below x kpa) so effectively another option 'active below kpa' in afr/o2 option. I wouldn't like to use closed loop fueling all the way to redline at high boost unless the mega can alter and maintain in timely fast manor?

Admanirv123, 2023-09-27:
Sure I will change, I thought its was clear in the first post esp with max psi in o2/afr, I've had to justify it by more explanation.