speeduino/speeduino #1112: USB connection w/o TS kills engine

myk7777, 2023-10-07:
Laptop connection on the USB port kills the engine if Tunerstudio or another serial monitor is not running.  The outputs will stay off until TS is launched.  I’ve tried all variations of the reset control, custom 16u2 firmware, reset jumper is in place, scoped the reset pin and it’s held high continuously, tried removing 5v from the USB cable.

The only thing I’ve found that fixes this issue is to roll back to 202108 firmware, anything from 202201 till now has this issue.

myk7777, 2023-10-16:
I purchased another mega2560 board to rule that out and it behaves the same.  I also removed the Arduino from the v0.4 shield and fed the crank and cam signals straight to it from an ardustim while scoping the output pins and it still shows this same issue.  I changed the pin mapping over to the no2c and it’s the same.  Only thing I can find to fix it is rolling back to 202108.

zuripig, 2023-10-16:
Same here - tried all steps you've listed too, including a multiple megas 🤷🏽‍♂️

Admanirv123, 2023-10-16:
This appears to break connection to msdroid too, moreover the signature can not be recognised as there is a small window when it ill align with msdroid project vs ecu, this is the same as when flashing new firmware upon first connection you've got around 10-15 seconds.

Walterronny, 2024-07-13:
Ive been trying to find any fix for it and definitely it's about 16u2 firmware, also i made a custom boards for working with custom 16u2 speeduino firmware but nothing happens. tried many things replacing caps with different values etc and same problem. Never tested 202108 but maybe it's 16u2 firmware?

myk7777, 2024-10-21:
A year later…  Just tried this on the latest firmware and it still exhibits this issue.  I’m surprised this isn’t a bigger problem for people.  Having the ECU hang from just plugging the usb in seems like a major problem.  I’ve only played with Speeduino on the bench out of curiosity but this would be a no go for me in a car.  As before this problem doesn’t exist in 202108 and prior.

noisymime, 2024-10-22:
So just to confirm the conditions you see this under:

1. You've got external power to the ECU
2. Engine is running
3. USB is not connected.

You connect USB to a computer where, as far as you know, no serial applications (TunerStudio etc) are running and the outputs stop? 

I've just tested under the above conditions with a standard v0.4 board/mega2560 and everything works fine. There is no interruption to the fuel and ignition outputs at all when the USB is connected. Obviously if I open a serial connection through TS or another application (And don't have reset control active) then it will restart, but with nothing running the ECU continues to work as expected. 

This is with the latest firmware.

Admanirv123, 2024-10-22:
> So just to confirm the conditions you see this under:
>
> 1. You've got external power to the ECU
> 2. Engine is running
> 3. USB is not connected.
>
> You connect USB to a computer where, as far as you know, [...]

This is pretty much as you explained but have reset control enabled (might have that wrong way round but don't want audrino resetting) as per the op and it cuts the engine if its running upon connection to TS.

Walterronny, 2024-10-22:
what's purpose to just connect usb as it, the connection to usb is for open or connect to TS. I tried many times to ask for help about this in Discord , one guy tried to help about it but had the same problems, it's not a reset problem it's like you connect usb and mega get block for a while then connect it, so advance reset control that is supposed to be for that doesnt work neither. i even uploaded videos, just only thing that appear to work was a firmware for 16u2 and iirc was about latency but can't remmeber , was ago long time about this problem.. decided to use always Bluetooth due USB could be crazy for new users..

> So just to confirm the conditions you see this under:
>
> 1. You've got external power to the ECU
> 2. Engine is running
> 3. USB is not connected.
>
> You connect USB to a computer where, as far as you know, [...]

myk7777, 2024-10-23:
> So just to confirm the conditions you see this under:
>
> 1. You've got external power to the ECU
> 2. Engine is running
> 3. USB is not connected.
>
> You connect USB to a computer where, as far as you know, [...]

Josh,

Yes to everything you've listed above.  Here's a video of the issue happening, also included my tune file.  For clarification, Tunerstudio is already running prior to the first USB connection in the video.  This is with the reset jumper in place and the latest firmware 202402.2.   Again this happens with all firmware since 202108.  Thanks for your time!

https://youtu.be/dXrMTkkx4-Q?si=J4QwdE5b_L_FDmxJ

Mike

[test1.zip](https://github.com/user-attachments/files/17484774/test1.zip)

noisymime, 2024-10-24:
OK, that video makes it clear.

My current thinking is that something on the PC is spitting out some data on the serial line when it's plugged in and this is triggering the ECU to wait for the rest of the command,. which obviously never comes. With TunerStudio running, it would trigger a connection reset and everything would be fine, but without it running the ECU is sitting there waiting for more data. 

It's just a guess, but that's the only thing I can immediately think of. 

I'll dig out a Windows machine and see if I can reproduce with that. What serial interface chip is on the arduino board you're using, the 16u2 or a CH340?

myk7777, 2024-10-24:
I’ve tried this with the two 2560 boards linked below, both have the 16U2 chip and both produce the same result.  Interesting hypothesis on the issue.  I’ve tried this on two separate win10 machines (laptop and desktop) with the same result.  They also both work fine with firmware 202108 and prior so it seems something changed there.  Thanks for digging into this!

https://a.co/d/eUUHskC

https://a.co/d/5kz1oTi

noisymime, 2024-10-24:
> They also both work fine with firmware 202108 and prior so it seems something changed there

There were some BIG changes between the `202108` and `202201` firmwares around serial comms, which is part of why I'm thinking it might be related to that. 

I'll try Win 10 and see if I can get the same thing happening there. If I can reproduce the issue I'll do a bisect and figure out exactly what caused it.

noisymime, 2024-10-24:
Wow, confirmed this on a Windows laptop! I'm really surprised as I hadn't seen this before.

The same board, running exactly the same firmware, works fine when connected the same laptop but running linux, works fine, so Windows must be doing something weird here. I typically only use Mac and linux, so somehow this hadn't been picked up. 

I'll do some digging and hopefully find a fix.

myk7777, 2024-10-25:
Glad you were able to replicate the issue.  I’m sure you’ll be able figure it out.  Will be interesting to see if the reset control features work correctly once this is sorted, from what I remember I was trying to figure that out when this all started.  Thanks again for your effort!

noisymime, 2024-10-25:
OK found it and submitted ba77e36 to fix. 

This was an interesting one. Essentially Windows is changing the DTR line on the serial port when the USB is plugged in, which triggers value `0xF0` (240) to be sent to the arduino. This is correct per the standard UART protocol whenever the DTR line is changed, but it appears that only Windows is doing this DTR change when you plug in, not linux or Mac. 

The firmware was then treating this 0xF0 as the first byte of the size field (The size field is the first 2 bytes of a serial packet) and then waiting for the rest of the message, which never came. 

Ordinarily this won't matter because when you plug the USB in you connect to TunerStudio which causes a reset anyway, but if you're not actually opening a serial connection then this issue was seen. 

Curiously, this was actually somewhat resolved already in the master firmware where a more robust timeout system has been added. However this meant that the system would only recover after the timeout period (Currently 700ms).  The fix I've added simply ignores this specific value so there is no interruption at all. 

Please try the current Dev firmware and see if this is resolved at your end as well, I'm reasonably confident it should be good now. I'll be adding this fix to the 202402 fixes branch and will include in 202402.3 also.

closed 2024-10-25 by noisymime

myk7777, 2024-10-25:
> OK found it and submitted [ba77e36](https://github.com/speeduino/speeduino/commit/ba77e36bd7400b8704945819fd3f14f71d6831dd) to fix.
>
> This was an interesting one. Essentially Windows is changing the [...]

Josh,

I compiled and loaded the 202402 fixes branch and it indeed fixes this issue, great job.  However it seems to have broken the ability to reconnect to TS via USB with the reset jumper in place?

noisymime, 2024-10-25:
> However it seems to have broken the ability to reconnect to TS via USB with the reset jumper in place?

Hmmm that's surprising as really this shouldn't impact the connection at all all. 

What method are you using with the reset jumper, is it simply a capacitor to the reset pin? Alternatively if you're not sure, what board are you testing with?

noisymime, 2024-10-25:
Actually I did find 1 small thing and have pushed up another commit to fix that. 

Can you try again with the latest master?

ric355, 2024-10-25:
For what it's worth, this is the 240 problem I had previously raised a couple of months ago. It isn't caused by Windows. It's caused by the firmware on the 16U2 reacting to the change in line speed when Windows (or Linux) adjusts the baud rate. The 16U2 firwmware goes through a UART reset process when the comms parameters are changed (e.g. setting to 115200 baud) and this in turn causes a blip on the serial lines between the 16U2 and the AtMega.  That blip results in a value of 240 being received by the AtMega. It won't necessarily happen all of the time because often device drivers will cache the baud rate and only change it if it is different to what is already configured. This is why it is not always seen.

I did not previously propose fixing it by rejecting the 240 value because hard coding one specific value is not guaranteed to fix it in all scenarios, since the value is generated due to reset noise on the TX line. It's very consistent though and the value is not one we use for anything else so we can get away with it despite it being a bit of a bodge.

The offending code in the 16U2 firmware is here:
https://github.com/arduino/ArduinoCore-avr/blob/master/firmwares/atmegaxxu2/arduino-usbserial/Arduino-usbserial.c#L204

And the originating post (relating to 240's being received in the serial monitor, because the serial monitor was continuously setting the baud rate when it didn't need to, thus continually causing the 16U2 to reset every 5 seconds or so):
https://github.com/arduino/arduino-ide/issues/375

Note that the originating post only fixes one specific application, and it not really helpful to us, since the underlying issue is the 16U2 firmware.

myk7777, 2024-10-25:
> > However it seems to have broken the ability to reconnect to TS via USB with the reset jumper in place?
> 
> Hmmm that's surprising as really this shouldn't impact the connection at all all. It's only discarding a byte that shouldn't be there in the first place.
> 
> What method are you using with the reset jumper, is it simply a capacitor to the reset pin? Alternatively if you're not sure, what board are you testing with?

The board I'm testing with is v0.4.3d from WTMtronics also using their component kit, was 2021 timeframe

myk7777, 2024-10-25:
> Actually I did find 1 small thing and have pushed up another commit to fix that.
> 
> Can you try again with the latest master?

Yes, I can confirm this fixes the issue on my end, nice work!  

I tested this will all 4 variants of the software reset control with the jumper on and off.  What I found is the software setting for reset control makes no difference in the behavior of the board with the exception of the reset lock dash indicator.  In all cases with the jumper in place it will never reset and stays running through USB disconnect and reconnect.  With the jumper removed regardless of the software setting for reset control the board always does a reset for ~1s while plugging in the USB.  Not sure if this is the intended functionality but I don't mind removing the jumper so the software setting option isn't a huge deal for me.  I also can't remember where I left off with the 16u2 firmware from a year ago so I'd need to dig into that to give a better answer.

Thanks for fixing this!

myk7777, 2024-10-25:
> For what it's worth, this is the 240 problem I had previously raised a couple of months ago. It isn't caused by Windows. It's caused by the firmware on the 16U2 reacting to the change in line speed when Windows (or Linux) adjusts the baud rate. It _will_ occur on Linux if you get the conditions right. The 16U2 firwmware goes through a UART reset process when the comms parameters are changed (e.g. setting to 115200 baud) and this in turn causes a blip on the serial lines between the 16U2 and the AtMega. That blip results in a value of 240 being received by the AtMega. It won't necessarily happen all of the time because often device drivers will cache the baud rate and only change it if it is different to what is already configured. This is why it is not always seen.
> 
> I did not previously propose fixing it by rejecting the 240 value because hard coding one specific value is not guaranteed to fix it in all scenarios, since the value is generated due to reset noise on the TX line. It's very consistent though and the value is not one we use for anything else so we can get away with it despite it being a bit of a bodge.
> 
> The offending code in the 16U2 firmware is here: https://github.com/arduino/ArduinoCore-avr/blob/master/firmwares/atmegaxxu2/arduino-usbserial/Arduino-usbserial.c#L204
> 
> And the originating post (relating to 240's being received in the serial monitor, because the serial monitor was continuously setting the baud rate when it didn't need to, thus continually causing the 16U2 to reset every 5 seconds or so): [arduino/arduino-ide#375](https://github.com/arduino/arduino-ide/issues/375)
> 
> Note that the originating post only fixes one specific application, and is not really helpful to us, since the underlying issue is the 16U2 firmware.

This surpasses my level of knowledge on the topic so I'll let you guys sort this out, I'm happy to test though.  Thanks!

noisymime, 2024-10-25:
> For what it's worth, this is the 240 problem I had previously raised a couple of months ago. It isn't caused by Windows. It's caused by the firmware on the 16U2 reacting to the change in line speed when Windows (or Linux) adjusts the baud rate.

I did have a recollection that something like this may have come up previously, but hadn't connected the dots around them. 

I did go looking in the 16u2 firmware code to see if there was anything that would specifically generate the 0xF0 / 240 value as this was the only one 1 ever saw in my testing (And seems to be the case for others as well), but I couldn't (And still can't) see why that is the value that gets sent. 

> https://github.com/arduino/arduino-ide/issues/375

Yeah I saw this as well, but it's specific to the arduino IDE serial monitor rather than trying to resolve the root cause of why this value gets sent in the first place. There's a couple of other odd mentions of this for 16u2 based boards around the place, but I'm surprised it isn't considered a bigger issue. 

Either way the fix that's in should be very safe. The 0xF0 value will never valid (It would mean a serial transmission of over 60kb in a single frame) so simply ignoring it won't cause any issues. It's crude, but it works.

ric355, 2024-10-26:
> I did go looking in the 16u2 firmware code to see if there was anything that would specifically generate the 0xF0 / 240 value as this was the only one 1 ever saw in my testing (And seems to be the case for others as well), but I couldn't (And still can't) see why that is the value that gets sent.

I also went looking for values of 0xF0 but I soon realised this is the wrong thing to do. It's also why it's hard to see how that value originates. It's a hardware reset of the UART on the 16U2. During this reset the outputs stay on, and they are not electrically isolated from the TX/RX on the ATMega. Therefore any noise on those lines during the reset will register as bits within the UART on the ATMega. This is why the value is not "random": the ATMega's UART clock is already set, and the reset process in the 16U2 UART is a fixed process with a finite timing.  I have had a logic analyser on the TX/RX pins (which are also the output of the 16U2's UART) and observed directly this value coming from the 16U2 UART while directly controlling what data is going over the USB Serial line from a terminal application. So I know 100% that the 16U2 is the originator.

Yes, as I said, the Arduino IDE fix is not useful to us. It's just another workaround. But the reason I linked the thread is it was the route I took into finding what actual the root cause is.

The main reason this issue is not affecting many people is because, as far as I've been able to gather, almost no-one is using reset control. Without reset control the bad value of 240 is completely masked because it is received while the ATMega is going through a reset itself. I have a suspicion that this is a self-fulfilling prophecy; nobody uses reset control because this issue means that reset control doesn't seem to work, since the engine keeps stopping despite reset control being enabled !

As a side point, it's possible to directly control whether the Arduino will reset when you connect by writing code that does not toggle the DTR status (over the USB, since there is no actual wire) when you connect. This is how my dash software works when it connects over USB, so it doesn't actually need reset control at all. It just so happens that TS and many other terminal applications automatically toggle DTR ("Data Terminal Ready") either because they think they should or because they actually want to force a reboot of the ATMega (Arduino IDE  for example).

dantob, 2024-10-27:
If you add a 10kΩ pullup to 5V between the 16U2 and arduino RX (PE0) does it help? It should force the UART line to idle while the 16U2 resets, unless it's purposely pulling it low for some reason. Actually, it looks like you can just use the D0 header.

ric355, 2024-10-27:
I don't understand whether that helps or not to be honest, but I don't think we can fix issues in the software by asking everyone to modify their hardware.