speeduino/speeduino #1126: Add sequential capability to daihatsu+1

runesmdk, 2023-10-30:
Trigger contains both speed and home signal.

Possible to make it sequentially capable? 

Would love to do away with my old rotorbutton and rotor cap :)

noisymime, 2024-01-18:
I can't see any reason why this won't just work if the sequential flag is turned on. It's already tracking crank angle over 720 degrees, so it should really be that simple. I'll give it a try and see if it simply works already

runesmdk, 2024-01-19:
> I can't see any reason why this won't just work if the sequential flag is turned on. It's already tracking crank angle over 720 degrees, so it should really be that simple. I'll give it a try and see if it simply works already

Love it, cant wait to try it

commit fdb6e492 (2024-01-22): Enable Daihatsu+1 decoder for sequential. Tested on 3 cylinder configuration

Fixes #1126
closed 2024-01-22 by noisymime