speeduino/speeduino #1129: Open/close angle injection timing selection

DrFiamma, 2023-11-04:
It would be very usefull if on TunerStudio the user could select open or closed injection timing based on the same 4x1 table.

Thanks!

noisymime, 2023-11-05:
This should be very simple, it's actually less work than work happens now as there's not a separate calculation needed for the start angle. All that's needed is to add the option and a a check. 

I'm curious what the benefit of this is though as typically you want to align with the point that all the fuel is delivered rather than when it starts getting delivered. Is there a specific use case for this?

cano507s, 2024-05-21:
It may be able to be used to spray fuel while the valves are opening. It would be easier to synchronize by changing the opening angle and not the closing one.

spburgess23, 2024-07-01:
This would be very useful for older engines with Siamese inlets. Not many around but having an affordable and available ECU opens many doors in the classic car world.

TakumiPT, 2024-10-20:
Hi, @DrFiamma  need to talk to you about your Clio with bosch Monopoint. I have some questions i am converting my Passat 1984 1.6 from carburetor (pierburg 2e2) to the same Monopoint. Found your tune. https://tunes.speeduino.com/#/t/BXdoyTuwrG

kovama1, 2025-02-28:
Hi! 
We have just finished fuel injection conversion for classic MG MBG 1800, and as I just learned we are facing issue with the Siamese Port Injection fuel distribution across cylinders. 
Without abbility to set injector opening angle we are unable to inject fuel in correct way (after 30-40° from TDC => needs to be after cyl 1-2 and cyl 3-4 valve overlap) resulting in very poor runnig engine. 

It would be wery helpful to us if anybody could take a look at this function. I am skilled electronic design engineer and I have a scope  thus I can verify exact injector opening times so I will happily assist with verification of this added functionality. 

[image]

Thank you for helping out, this is only way how to get our classic MG running!

kovama1, 2025-03-01:
An just to give more technical info on our setup. 
We have full sequential setup (both 36-2 crank and one sensor instead of original distributor). Injectors are wired in the way so can fire injector in 1-2-2-1 sequence. Injectors used are EV14 BOSCH 0 280 158 040 - 980cc => this should be hopefully enough to supply fuel for our little turbo toy :D

[image]

Most valuable picture is this one (source: [https://www.starchak.ca/efi/siamese.htm](url))
We can se that for engine with siamese head we have only limited time to deliver fuel to outside cylinder (blue parts). We need to start injecting fuel after inside intake valve is almost fully closed (around 30° depending on actual cam profile), and stop injecting fuel before outside intake valve closes. Any fuel injected after closing outside valve will be sucked in with inside cylinder and thus making it rich.

[image]

To start with this just simple setting for injection start angle would allow this engine to at least run at light loads. Since we have turbo we would need to limit also injector closing angle to avoid charge imbalance at higher duty cycles (consequently limiting max injector duty cycle).

Hope this helps to understand this more ;)

spburgess23, 2025-09-27:
@kovama1 I have spent many hours looking at fuel injection for the MGB, we are fairly well known in the B world for cylinder head work and engines etc. a couple of years ago I modified the code to allow manual injection control across two independant injectors which worked quite well. 

a rough video but you get the idea...
https://www.youtube.com/shorts/F2IEYuPdxLU

The other thing to watch out for with an MGB engine is the world of charge robbing, you have at one point 1.5 valves open in the inlet ports. and a bizarre occurrence happens. At low speed / off cam cylinder 1,4 rob the charge. At high speed / on cam cylinders 2,3 rob the charge, the crossover happens at a unique point with every engine so have to be tuned on a per engine basis, so a standard cam would swap charge robbing at around 1000-2000RPM vs a piper BP300 say would have the siamese port from happen from 3500-4500RPM. so in order to get the injection angles correct we found you need to have modern very fine injectors, fire on two closed valves for the pre charge rob, the the charge robbing cylinder you have to inject just as the second valve starts opening to get adequate fuel into the mix. 

I have lots of notes and logs from egt/lambda sensors somewhere, the efi project has been very much on the backburner because the only after marked ecu that could handle maps within maps has been taken off the market, however I reluctant to give up on a simpler way of doing the injection and hope that with time the speeduino project allows this to happen. the ethos of the project is very much inline with our ethics in the MGB tuning world.