speeduino/speeduino #1132: 3 cyl wasted cop vs 6cyl wasted spark

runesmdk, 2023-11-06:
3cyl wasted cop doesnt run right, a cylinder is missing. Tried reversing coil 2-3, but that just made it worse.  Setting it as 6cyl wasted spark it runs right, with original coil wiring too.

runesmdk, 2023-11-06:
Added: 202305 firmware

noisymime, 2023-11-06:
Is one of the 3 spark outputs not firing at all, does it fire a the wrong time? Wasted COP on a 3 cylinder is the same as just normal wasted spark isn't it?

runesmdk, 2023-11-06:
All 3 fire, timing is off.
Talked with pasi earlier. He said the description of said modes arent correct, and it should work with wasted spark (not cop) in 3cyl setting as well.

So seems to be a description issue. 

Car runs for now :)

noisymime, 2023-11-06:
OK, I see the problem here. There's an init case for Wasted COP that hasn't considered 3 cylinder. Fix coming shortly

commit dd76cfa0 (2023-11-06): Change init for <=3 cylinder Wasted COP to match standard wasted spark

Fixes #1132
closed 2023-11-06 by noisymime