speeduino/speeduino #1135: Dropbear secondary serial does not work

paulgilc, 2023-11-11:
Known issue with secondary serial on Dropbear.  Code points to Serial2 but should be Serial1 to enable the use of the J4 header.  Just setting to Serial1 in the secondary serial code does not work as has been advised and instructed to do.  If you set it to Serial6, and then use the pins on the Teensy for Tx6-Rx6, it then works.

paulgilc, 2023-11-11:
I can confirm that changing the below in "comms_secondary.cpp". Current Master Code.  Works.
Setting it to Serial1 does not.
#elif defined(CORE_TEENSY)
  HardwareSerial &secondarySerial = Serial6;

noisymime, 2023-11-21:
This should be resolved in master now. Please test and let me know if you're still seeing issues.

commit 36d22b48 (2023-11-21): Allow secondary serial interface to be set on a per board basis rather than MCU only

Fixes #1135
closed 2023-11-21 by noisymime