speeduino/speeduino #1136: more efficient 3d tables?

Fera31, 2023-11-13:
there is the possibility of adding more efficient tables. for more powerful hardware?
for example 0.1%
Or would I have to change the entire 3D table section?
for example the can bus option - it is only enabled if the hardware allows it
Of course, tuner studio will also have to be modified to accept these tables.

dynfer, 2023-11-13:
Depends purely on the used hardware the Mega is limited by the flash size while on a Teensy / STM You can do that freely.

noisymime, 2023-11-13:
This isn't so much a flash size or a processing problem, it's a storage one. 

The EEPROM/non-volatile storage on most of these boards is still 4KB and that is fairly limited with the current tunes. Increasing the resolution of a 16x16 table requires an additional 256 bytes per table. It *may* be feasible to do this on some selected tables (Fuel and/or ignition potentially) but beyond this it would be unlikely without a universal way to add additional storage.

Fera31, 2023-11-13:
Sure - but in stm32f4 that would be very feasible. And it wouldn't cause problems, right? I'm going to investigate how to do it. since I have no idea how
I think the system works very well. But it's time to upgrade to more powerful hardware.

adbancroft, 2023-12-24:
> I would appreciate if someone can guide me

Seems like a good way to get feedback on the code documentation :smile:. Start here: https://speeduino.github.io/speeduino-doxygen/group__table__3d.html. Have a read, ask questions on Discord.

If you are just trying to increase the size of the tables, this has been discussed a few times: in code it's doable, but gets complicated in the INI file.

adbancroft, 2024-01-20:
Reading the [INI file manual](https://www.efianalytics.com/TunerStudio/docs/EFI%20Analytics%20ECU%20Definition%20files.pdf), there is a section on dynamically sized arrays. This may help you: the users chooses their target board in the tune, so you could adjust table sizes based on the chosen board?

(I haven't tested this, just a thought experiment!)

Fera31, 2024-01-30:
> Reading the [INI file manual](https://www.efianalytics.com/TunerStudio/docs/EFI%20Analytics%20ECU%20Definition%20files.pdf)[image], there is a section on dynamically sized arrays. This may help you: [...]

I read the file and I understand it perfectly, it works like a motec m800. It would also be good to be able to select the axes. like rpm map tps clt etc. but the truth is that my brain can't do it, I understand how it works but I can't put it into code.
anyway thank you very much

adbancroft, 2024-01-30:
@noisymime Recommend we close this one for the moment.