speeduino/speeduino #1145: Sequential not working without cam signal.

TBAMax, 2023-12-09:
In the discord chat user galicognjen pointed out that it is possible to make the sequential work without cam signal. Only by detecting if the engine starts and swiching the rotation if it do not start with the first try (need to get it right if it is rotation 1 or rotation2 of the engine cycle for 4 stroke engine).

Kind of cheapie approach skimping on fitting a cam sensor, yet this is usually a pain to find the good place for the cam sensor, and usually it is really hurry to get it running quickly.

Looks like this is not implemented yet but seems would be really nice to have...

smclt30p, 2023-12-09:
Hi, I'm that user. Following.

MykkH, 2023-12-12:
I'm not sure what the issue or problem you are trying to identify is, however I hope you arent requesting that the sequential only be functional when a crank and cam signal is present, my next install will be using a single 60-2 at cam speed for sequential operation.

noisymime, 2024-03-14:
I understand the theory of this approach, detecting non-firing on one phase and then trying on the next, however from testing this is an extremely difficult function to get work in a generic way (ie not specific to 1 engine). I don't believe the audience for this is large enough to justify the considerable time and effort required to get it right. 

Closing for now unless someone volunteers to tackle this one.

closed 2024-03-14 by noisymime