speeduino/speeduino #1147: Suggestion: Remove current references to SAM D21 & E51

FoxUnpopuli, 2023-12-12:
Purely selfish, as I'm trying to add in support for the SAM D51P20A/E54P20A myself.

I would (attempt to) run through the code, removing all references and unnnecessary libraries for the smaller Atmel/Microchip SAM microcontrollers.

Autohome2, 2023-12-12:
Just curious why is this necessary if support is being added? Confusing

noisymime, 2023-12-13:
Whilst the existing files for the D21 and E51 boards are incomplete and currently non-functional, I can't see why they would get in your way. Just create a new board file for the platform you're targeting (Eg D51) and ignore the ones that are already there.

FoxUnpopuli, 2023-12-15:
Development is well under way.  My concern was clarity and tidiness.  It would be nice to know which dev boards were being used for the D21 / E51 - although I was trying to focus on the D51/E54 without being distracted, it might be more sensible to try to finish the work others started?

Autohome2, 2023-12-15:
I think you are missing the point on what the board.h file should setup.
It should NOT be targeting a specific dev board.
The board file should configure the MCU specifics, then the ACTUAL board configuration is setup in the init file with the board TYPES.
This means only one board.h  file is needed for a particular MCU, and the user chooses or creates the board TYPE config for the design of their PCB or dev board

FoxUnpopuli, 2023-12-16:
Yes, I understood that, quite clearly and I went back and forth a couple of times on '**board_SAMD51**' and '**board_grandcentral**', choosing the latter because it's under my nose and thus the first success will be specific to the board before I can look at making it SAMD5x/E5x generic.  I have a couple of paths forwards if this pans out to more than an intellectual/bench-based coding exercise - one is a Teensy-footprint SAM E54 board, the other is a Dropbear derivative that suits the Grand Central/MEGA pin confinguration.  Having had a quick look at the Arduino UNO R4, that chip has some nice stuff under the hood also, and plenty of timers - using a Dropbear architecture concept with the UNO R4 could work... but I have digressed.

If I can get somethink working, yes, I will convert from Grand Central to SAMD51P20A/SAME54P20A, and maybe some of the other SAMD5x/E5x chips, **if they have suitable dev boards available and people ask for them.**  If you're willing to help me, I'm all for it, my fork 'master-addgc' is [in my Github account.](https://github.com/FoxUnpopuli/speeduino)  Don't be too harsh, I am 'inexperienced' at best, and 'totesn00b' at worst.  It has been 20 years since I last did proper C/C++ in anger, my learning curve has been steep for the past couple of months.

Right, where are the **board_MK64FX512.*** and **board_MIMXRT1062.*** files?  ;)

-Mark.

noisymime, 2024-03-15:
Closing this as no action is required. If support for D51 MCUs is added it should simply be in a `board_D51.x` file. 

> Right, where are the board_MK64FX512.* and board_MIMXRT1062.* files? ;)

Thee are slightly different as these MCUs are not supported in isolation, only in combination with the Teensy bootloader and microcode.

closed 2024-03-15 by noisymime