speeduino/speeduino #1150: 4-1

Fera31, 2023-12-16:
i have 4-1 trigger in cam.  all sequential 4 cylinders .the ecu took 3 to 5 crank turns in start to count rpm. 
how can I solve that ?

digmorepaka, 2023-12-17:
If you're running cam primary, you will need at least 2 revolutions of the crank to get sync. Make sure your sensor gap is also correct.

Fera31, 2023-12-17:
> If you're running cam primary, you will need at least 2 revolutions of the crank to get sync. Make sure your sensor gap is also correct.

yes. the car runs perfect. i have 4 cars runing in the same pattern . but is very irritating because the starter is slow

noisymime, 2024-01-12:
As mentioned, it will take a minimum of 2 crank revolutions, but you can also look at things like the Skip Revolutions setting to try and speed up the starting process. What do you have that set to?

closed 2024-03-15 by noisymime