speeduino/speeduino #1154: Bug: secondary spark table applies ignition corrections in multiply mode

adbancroft, 2023-12-24:
In multiply mode, we treat the values in the second spark table as percentages: the INI specifies a range of 0-215.

But `getAdvance2()` applies a couple of changes to the table value *before* it's used as a percentage:

1. Ignition offset - potentially making the value negative
2. Ignition advance corrections - which are degree based corrections.

Also worth noting that statuses::advance2 is int8_t, which doesn't have the range to store the range defined in the INI.

adbancroft, 2023-12-24:
PR incoming...

noisymime, 2024-02-20:
I've pushed up a quick fix for 2. making it so that the advance corrections are only performed when a switched mode is used. 

This doesn't fix the potential for a negative value, however I believe that can only occur in added mode if someone puts a negative value in the table. In multiply mode the smallest value you can get after subtracting `OFFSET_IGNITION` would be 0

noisymime, 2024-02-20:
The added mode potential negative is addressed in #1155 and will be incorporated from there. I believe the changes of that occurring are fairly small.

commit 7361b461 (2024-02-20): Only calculate corrections values on the secondary spark looked if either of the table switching modes are used (Not Multiple or Add)

Fixes #1154
PR #1155 (2025-01-09): Bug fixes for getAdvance1, getAdvance2 & secondary tables.
Fixes #1152 
Fixes #1153
Fixes #1154 
Fixes #1274 

Details:
In addition to the bug fixes:

1. Add **full unit tests** for `calculateSecondaryFuel()` & `calculateSecondarySpark()`
2. Unit testability: split out the definition of statuses, page structs & bit fiddling from globals.h into separate header files (statuses.h, config_pages, bit_manip.h). 
- This allows code to have a dependency on the struct definitions without pulling in global state. 
- In turn, this allows code to have the current status & page structs injected. 
- Which a [... cut]
closed 2024-02-20 by noisymime