speeduino/speeduino #1156: Bug: analog sensor ISR conditional compilation uses wrong preprocessor symbol(s)

adbancroft, 2023-12-28:
In sensor.cpp, `AnChannel[]` is conditionally defined & declared using the `ANALOG_ISR` symbol. However, the code that uses `AnChannel[]` is conditionally compiled using _one of _2__ _symbols_:

- `ANALOG_ISR` (E.g. `readTPS()`); or
- `ANALOG_ISR_MAP` (E.g. `readBaro()`).

Shouldn't they all use the same symbol? (`ANALOG_ISR`)

adbancroft, 2023-12-28:
PR incoming

noisymime, 2023-12-29:
The intent of having separate flags (`ANALOG_ISR` and `ANALOG_ISR_MAP`) was to be able to test the MAP reading separately. It really needs to be read as often as possible for the different sampling modes, so it is handy to be able to test with the ADC turned on for everything except MAP. 

I don't think you would ever want to have this separation if this feature is ever turned on, but it is handy for testing.

adbancroft, 2023-12-31:
So you are setting ANALOG_ISR and varying ANALOG_ISR_MAP for dev purposes?

noisymime, 2023-12-31:
Yes, partially to test the MAP read response impact and partially to see whether the 2 modes could operate alongside one another (They currently can't). It's not a big thing that needs to be retained though, just mentioned the background as to why it was there in the first place.

adbancroft, 2024-01-02:
So not a bug. 

I'll close this issue but leave the PR open - it has some useful changes in it (MISRA fixes, a tiny reduction in RAM used).

closed 2024-01-05 by adbancroft