speeduino/speeduino #1163: OddSquirt not working on low RPM`s

RempageR1, 2024-01-09:
Hi,

I have a Vmax that will run it`s injectors close to cylinder head as possible. I do this by setting the injector degrees to the oddfire degrees as shown below:

channel2InjDegrees = configPage2.oddfire2; 
channel3InjDegrees = configPage2.oddfire3; 
channel4InjDegrees = configPage2.oddfire4; 

And this works just fine, until you get under 450 RPM. Then, injectors 3 and 4 go nuts and fire about twice as much as 1 and 2 do. Above 450 RPM everything is well.

In my setup, channel 2 is at 180 degrees, which is the same as even fire, so that explains why 1 and 2 are fine. 3 is at 430 and 4 is at 540.

Is there any way to fix the oddSquirt behavior at low RPM? Cranking is set to 400, so it`s not that.

Thank you.

RempageR1, 2024-01-10:
Connected the scope today and the issue is also there on channels 1 and 2 (albeit less). So this is a generic issue on low-RPM fuel delivery.

RempageR1, 2024-01-11:
Did some more testing, and you can see the scheduler status is different for output 3 is different at the time I see the issue. The two Serial.print() commands were added to all 4 injectors:

-----------------------------------------------
#if INJ_CHANNELS >= 1
      if( (maxInjOutputs >= 1) && (currentStatus.PW1 >= inj_opentime_uS) && (BIT_CHECK(fuelChannelsOn, INJ1_CMD_BIT)) )
      {
        uint32_t timeOut = calculateInjectorTimeout(fuelSchedule1, channel1InjDegrees, injector1StartAngle, crankAngle);
        if (timeOut>0U)
        {
Serial.print("1 ");
Serial.println(fuelSchedule1.Status);
        setFuelSchedule(fuelSchedule1, 
                      timeOut,
                      (unsigned long)currentStatus.PW1
                      );
        }
      }
#endif
-----------------------------------------------

And the one that bliks twice is #3.
1 0
2 0
3 1
4 0
1 0
2 0
3 1
4 0
1 0
2 0
3 1
4 0

I really hope there is someone that can work with this because it`s above my coding stills. You can reproduce this in the latest dev release and 60-2 trigger wheel using even-fire. Then try some RPM`s below 400 and you`ll run into the same issue.

adbancroft, 2024-01-11:
> You can reproduce this in the latest dev release and 60-2 trigger wheel using even-fire. Then try some RPMs below 400

On a Mega, I setup 60-2 missing tooth crank speed wheel, used the oddfire settings you describe (0, 180, 430, 540), cranking RPM at 200 and used Ardustim to feed a 350 RPM signal. A logic analyzer hooked to the Mega shows a nice even set of injector traces:
[image]

Latest master.

What setting am I missing to repro?'
[image]

[image]

RempageR1, 2024-01-11:
It only happens at very specific RPM`s like 348. But that number is just a guess. I can get you the exact number tomorrow and I`ll use your exact settings then. I use the Teensy3.5 so that might be a difference. This is what my scope shows, Ignite 3 vs Fuel 3:

[image]

noisymime, 2024-01-11:
Can you post your full tune please. 

I cannot reproduce this at a quick try, testing between 250 and 500 RPM I get results as expected on all 4 outputs. This is testing with a mega2560 though as I didn't have the Teensy set up. It is possible that this is someething to do with the lower overall timer timeout on the Teensy, but I can't see why that would only impact a certain channel and is getting checked for anyway, so that seems unlikely. 

I'll try with your full tune though.

RempageR1, 2024-01-12:
[2024-01-12_15.57.19.zip](https://github.com/noisymime/speeduino/files/13921141/2024-01-12_15.57.19.zip)

Above is the standard tune I use to reproduce this. I see the issue at 256 RPM on the Teensy. I`ve also dusted off my old mega2560 and indeed the issue isn't there. If you need me to test something with the Teensy I`d be happy to help...

adbancroft, 2024-01-12:
You are using VMax, right?

I noticed this in `triggerPri_Vmax()`:
  `setFilter((curGap/1.75));//Angle to this tooth is 70, next is in 40, compensating.`

`curGap` is `unsigned long` & the `setFilter` parameter is also `unsigned long`.....so this doesn't do what the author thinks it's doing.

Possible issue, though the code has been there for almost 2 years.

adbancroft, 2024-01-12:
> As I`m not really a coder, perhaps you can tell me why this is a problem so I can fix it.

I *thought* it would apply integer division, but after writing a [quick test](https://godbolt.org/z/qxn1fz6P6), it looks like it might be doing what's expected after all - at least on a desktop compiler, the divisor is promoted to double, followed by a floating point divide and then truncation to unsigned long. Which is probably good enough

It will be horribly slow on anything that doesn't have an FPU though.

RempageR1, 2024-01-13:
Thank you for speeding up this code :).

If there is anything I can test for the timing issue on the Teensy i`d be happy to help. Please not it only occurs in the injection channels, and not the ignition channels.

giovannni97, 2024-01-17:
Hi, I also am using the Vmax decoder but in order to really get it working i needed to change the MAX9924 VR interface from A2 mode to C --> Our "large tooth" is otherwise often misinterpreted because it trigger on the zero crossing which might be just noise and not the falling edge of the tooth.

mike501, 2024-01-17:
I've seen this issue mentioned a number of times with the max VR
conditioner. Do have a rough rule of thumb when it becomes an issue that
can be added to the wiki as a general guide, eg trigger wheels with less
than x teeth etc etc

giovannni97, 2024-01-17:
The MAX VR conditioner "power" lies in its A2 mode which is essentially an automagically adjustable hysteresis filter so that each positive trigger is recognized if it is at least 1/3 of the previous peak amplitude. The negative peak tho is basically ignored since when the signal goes to zero even for an instant the output goes low, no filter added.
This is generally a non problem since 99% of trigger wheels have fixed-width teeth and the position is given trough gaps --> the system only need to care about either the start or the end of a tooth, not togheder in the same application.
My Yamaha XTZ750 and @RempageR1 though have differently sized teeth and the position is obtained by looking at each lenght --> we need to look at the start (positive pulse) and end (negative pulse) on each tooth.
The MAX VR on A2 mode will basically be always spot on for the positive trigger but almost always mess up on the negative.
The barebone method is to add a low freq LP filter (high R and C on input of MAX VR) to hope that the residual voltage on the CAP will prevent the signal from crossing to zero after the poisitive pulse until the negative (end of tooth) pulse happen.

The mode C which I adopted loses the automatic filter on input (now is fixed value for the HYS threshold) but it is now symmetrical so that I can have a reasonable RC on input and see both rising and falling edge when needed.

Sorry for the long post but it is a complex issue not fully discussed given the limited times it is truly needed

RempageR1, 2024-02-12:
I`ll close this one and open one with proper (new) information. This is happening on both Teensy 3.5 and my new 4.1 setup. It happens on all injection channels, all modes not only oddifre but also in paired or semi-sequential mode.

closed 2024-02-12 by RempageR1