speeduino/speeduino #1166: Air Conditioning Control issue

nuosu, 2024-01-22:
In the Air Conditioning Control function, the idle up feature does not work in ” OLCL STEPPER mode“.

turboedge, 2024-01-22:
Can confirm this does not work in open loop either. Step gauge does not change when AC input changes. It would also be helpful if the change field could accept negative values as I run my table backward, eg more steps=more closed=slower idle.

turboedge, 2024-02-12:
> 
Well I feel silly. It would seem the firmware that I had was older than I thought, and this does work in open-loop stepper mode now. Apologies

noisymime, 2024-02-19:
This appears to be working fine in my testing. I am seeing the IAC Load value increasing by the expected amount when the A/C Request Input goes active. 

Can you post your tune and I see if there's anything that stands out, but this does appear to be operating as intended. What version of the code are you testing with?

noisymime, 2024-05-30:
Closing this as I cannot reproduce the issue. Please feel free to re-open with a log/tune if you're still seeing it.

closed 2024-05-30 by noisymime