speeduino/speeduino #1181: Dual injections on low RPM with Teensy 3.5 and 4.1

RempageR1, 2024-02-13:
Hi,

I am opening this issue as a continuation of #1163. There I thought there was an issue with oddsquirt but the issue is much bigger than that. 

In any mode (paired, semi-sequential, sequential) there are multiple injections below a certain RPM. With Teensy3.5, the issue occurs sporadically at certain RPM`s and goes away around 450 RPM. 
I also tested it with a Dropbear-like unit with Teensy 4.1 and MC33810s which also works fine up to around 450 RPM. Then, channel 4 starts to behave strangely (dual injections) sequential mode followed by channel 3. A bit higher in the RPM, channel 2 also shows dual injections and around 1000RPM all four channels have dual injections. At 1080RPM, suddenly all injector channels start to behave as they should. This only affects the injectors, NOT the ignition.

On the mega2560 the issue isn’t there.

Attached are five pictures. Blue is the injector, yellow is ignition of #1 for reference.
1 shows channel 4 @ 543 RPM
2 shows channel 4 @ 940 RPM
3 shows channel 3 @ 940 RPM
4 shows channel 1 @ 940 RPM
5 shows channel 1 @ 1110 RPM.

[image]
[image]
[image]
[image]
[image]

I hope someone can have a look at this. If I need to test anything with the Teensy 3.5 or 4.1 please let me know.

RempageR1, 2024-02-15:
I did quite some testing with this one and came up with a workaround. It is not pretty and I am probaly fighting symptoms and not the root cause but perhaps it helps someone with the proper coding skills to identify the problem. Whenever I divide the timeout that is sent to the setFuelSchedule() function, the RPM`s where the issue occurs is divided by 2 as well. So if the issue occured at 1000 RPM, the timeout/2 makes it happen at 500, and no longer above. If I however divide the timeout too much, I am running into issues at the higher RPMs. The function below now has code that divides the timeout by 64 below 200RPM and divides by 8 below 1200RPM:

`
static inline uint32_t _calculateInjectorTimeout(const FuelSchedule &schedule, uint16_t openAngle, uint16_t crankAngle) {
  int16_t delta = openAngle - crankAngle;
  if (delta<0)
  {
    if ((schedule.Status == RUNNING) && (delta>-CRANK_ANGLE_MAX_INJ)) 
    { 
      // Guarenteed to be >0
      delta = delta + CRANK_ANGLE_MAX_INJ; 
    }
    else
    {
      return 0;
    }
  }
  if ( currentStatus.RPM < 200){
    return angleToTimeMicroSecPerDegree((uint16_t)delta)/64;
  }
  else if ( currentStatus.RPM < 1200){
    return angleToTimeMicroSecPerDegree((uint16_t)delta)/8;
  }
  else{
    return angleToTimeMicroSecPerDegree((uint16_t)delta);
  }  
  
  //return angleToTimeMicroSecPerDegree((uint16_t)delta);
}
`

noisymime, 2024-02-15:
Suspect I can see the cause of this. Will test later today and get a fix out asap if it is that

noisymime, 2024-02-15:
Strangely I could not reproduce this at all on either T3.5 or T4.1. I'm assuming your testing has been with the latest code on master or was this with one of the older released versions? 

I've pushed up a small change that aligns the logic fuel setting function exactly with the ignition one, though I would be surprised if this makes a difference. It also includes a small change on the T3.5 timeout value for safety. 

Could you please try the current dev code with that change pushed and see if you get the same issue?

RempageR1, 2024-02-16:
Yes, I am testing with the latest version. I`ve pulled the latest version and tested on my teensy 4.1.

Be prepared to be surprised because it does fix the dual injections :). I`ll also try and see if it fixes it on the 3.5, after that I think we can close this one.

Last year I was very confused why the O2 was all over the place at idle, but dual injections might just be the reason why :)

Thank you for fixing this!

RempageR1, 2024-02-16:
Teensy 3.5 also works fine now. I`ll close this one. Thanks again!

noisymime, 2024-02-16:
Well I won't complain if that's resolved it, but it's certainly surprising. Were you using one of the recent pulls from master or something from last year?

RempageR1, 2024-02-17:
Hi noisymime, yes I was using the master, current version. Not something from last year. If it works, it works :).

RempageR1, 2024-02-24:
Spoke too soon. I tested with the Injector timing at 0, which worked flawlessly. However if you set them to 360 (or above 200), I get dual pulses all all injectors at certain RPM`s. So no not fixed I`m afraid...

RempageR1, 2024-03-10:
Well, I fixed the remaining issue for my setup. The calculateInjectorStartAngle does get the currentStatus.injAngle as an argument, but the calculateInjectorTimeout() doesn't. I think it is therefore misaligned. What I did for now is the following:

I added this to the speeduino.ino, so it uses the injector degrees (oddsquirt in my case) plus the injector close angles at the beginning of the timing routine. Therefore the correct channelXInjDegrees are used for both calculateInjectorStartAngle() and calculateInjectorTimeout();

      //***********************************************************************************************
      //BEGIN INJECTION TIMING
      currentStatus.injAngle = table2D_getValue(&injectorAngleTable, currentStatus.RPMdiv100);
      if(currentStatus.injAngle > uint16_t(CRANK_ANGLE_MAX_INJ)) { currentStatus.injAngle = uint16_t(CRANK_ANGLE_MAX_INJ); }
      unsigned int PWdivTimerPerDegree = timeToAngleDegPerMicroSec(currentStatus.PW1); //How many crank degrees the calculated PW will take at the current speed

      channel1InjDegrees = currentStatus.injAngle;
      channel2InjDegrees = configPage2.oddfire2 + currentStatus.injAngle;
      if (channel2InjDegrees >= (uint16_t)CRANK_ANGLE_MAX_INJ ){
        channel2InjDegrees = channel2InjDegrees - (uint16_t)CRANK_ANGLE_MAX_INJ;
      }
      channel3InjDegrees = configPage2.oddfire3 + currentStatus.injAngle;
      if (channel3InjDegrees >= (uint16_t)CRANK_ANGLE_MAX_INJ ){
        channel3InjDegrees = channel3InjDegrees - (uint16_t)CRANK_ANGLE_MAX_INJ;
      }
      channel4InjDegrees = configPage2.oddfire4 + currentStatus.injAngle;
      if (channel4InjDegrees >= (uint16_t)CRANK_ANGLE_MAX_INJ ){
        channel4InjDegrees = channel4InjDegrees - (uint16_t)CRANK_ANGLE_MAX_INJ;
      }

Then, I removed the injAngle from the calculateInjectorStartAngle routine (it is now unused):
   
     ////int16_t startAngle = (uint16_t)injAngle + (uint16_t)injChannelDegrees;
     int16_t startAngle = (uint16_t)injChannelDegrees;

Hope this helps. If it looks good, I could create a pull request for this. Thank you.

RempageR1, 2024-03-11:
Update: Yep it can be done simpler. The issue is indeed the fact that calculateInjectorTimeout() doesn't take the injAngle value into account. Below shows the old and new calculateInjectorTimeout() routine. I added the injAngle from the currentStatus to the equation and ensure we do not go over the max degrees (360/720). This should be a permanent fix that works for all setups (showing old and new):

     /*
     static inline uint32_t calculateInjectorTimeout(const FuelSchedule &schedule, int channelInjDegrees, int openAngle, int      crankAngle)
     {
       if (channelInjDegrees==0) {
         return _calculateInjectorTimeout(schedule, openAngle, crankAngle);
       }
       return _calculateInjectorTimeout(schedule, _adjustToInjChannel(openAngle, channelInjDegrees), _adjustToInjChannel(crankAngle, channelInjDegrees));
     }
     */

     
     static inline uint32_t calculateInjectorTimeout(const FuelSchedule &schedule, int channelInjDegrees, int openAngle, int crankAngle)
     {
       int16_t startAngle = (uint16_t)currentStatus.injAngle + (uint16_t)channelInjDegrees;
       if (startAngle >= (uint16_t)CRANK_ANGLE_MAX_INJ ){
         startAngle = startAngle - (uint16_t)CRANK_ANGLE_MAX_INJ;
       }
       if (startAngle==0) {
         return _calculateInjectorTimeout(schedule, openAngle, crankAngle);
       }
       return _calculateInjectorTimeout(schedule, _adjustToInjChannel(openAngle, startAngle), _adjustToInjChannel(crankAngle, startAngle));
     }

Some remarks: 
- I used the currentStatus.injAngle directly in calculateInjectorTimeout() whilst it is being parsed as an argument to calculateInjectorStartAngle(). Perhaps it`s better to use the same method in both.
- Naming between calculateInjectorTimeout() and calculateInjectorStartAngle() is messy. In the first, the angle for the injector is called injChannelDegrees, in the latter it is called channelInjDegrees. It might be good to clean that up too. In speeduino.ino they are called channelXInjDegrees so perhaps this is the best name to use.

Please let me know if I should create a pull request for this. Thank you.

noisymime, 2024-04-05:
I've pushed up what should be a fix for this now. Fantastic work in tracking it down. 

Could you please verify that this current master is now working fine on your end?

noisymime, 2024-04-05:
I've rolled this back for now as I think I need to look at this logic a bit closer.

RempageR1, 2024-12-03:
> I've rolled this back for now as I think I need to look at this logic a bit closer.

Any updates on this? Would love to have this fixed so I do not have to patch it every time I pull a new release. Also I think this might affect quite a number of other Teensy users that might not even know it. When I was testing, the dual injections caused so much fuel to be added to 1 cylinder the crank reversed rotation on some starting attempts. That was the moment I stopped testing on the bike and got out the scope and Ardustim. It can be really dangerous for an engine to use it as-is.

Thank you.

noisymime, 2024-12-04:
I've looked at this on and off and I don't believe what you've called out is the issue here. 

The `openAngle` variable within the `calculateInjectorTimeout()` function already has the `injAngle` value included when it is calculated, so I do not believe that needs to be added in again as you've done. The changes you're making might be partially working around the issue to a degree, but causes the injector timing to be incorrect. 

I suspect there's actually 2 things going on and both need to be fixed in order for the issue to be completely fixed, however it's difficult to test as I cannot get this to occur more than once every 20s or so regardless of the conditions. I'm got some changes that I believe do solve this, but I need to test further. I will look to get these committed within the next day or 2 for testing.

RempageR1, 2024-12-04:
Hi noisymime,

Thank you for getting back to this. Since I was able to reproduce this 100% of the time, I`d be happy to do some testing for you and let you know if the issue is resolved. Just let me know if there`s anything I can do for you. Happy to assist :).

noisymime, 2024-12-04:
Give this branch a shot and see how it looks: https://github.com/speeduino/speeduino/tree/Issue-1181-schedule-rework

I still need to do quite a lot of testing as the changes are to some fairly deep parts of the scheduler, but it seems to be right here.

noisymime, 2024-12-04:
Ahhh sorry, I meant to say that this was only targeting the fuel issue at this point. The ignition side of things in that branch still needs aligning with the new fuel schedule code 😄

Good to hear the fuel issue seems right now though. I'll get the ignition side of things changed and tested as well and let you know. 

Thanks for testing!

RempageR1, 2024-12-05:
> Ahhh sorry, I meant to say that this was only targeting the fuel issue at this point. The ignition side of things in that branch still needs aligning with the new fuel schedule code, so not surprising it's broken. 😄
> 
> Good to hear the fuel issue seems right now though. I'll get the ignition side of things changed and tested as well and let you know.
> 
> Thanks for testing!

No worries! Happy to test and no time lost. Once you have a final one or so I can test, I`d be happy to help.

noisymime, 2024-12-06:
OK, I think the code in that branch should now be right to go if you're able to test. I can't see any issues with fuel or ignition pulses under any conditions now.

RempageR1, 2024-12-07:
Thank you, gave it a go. I spent quite some time testing and there still are some corner cases where the injection misses. On hindsight this was also the case in the first version you provided but I think I got a bit distracted by the Ignition. 

I`m running a bit of custom code where I tied the injector degrees to the oddifre config like this:
channel2InjDegrees = configPage2.oddfire2;
channel3InjDegrees = configPage2.oddfire3;
channel4InjDegrees = configPage2.oddfire4;

When setting 'injector characteristics' degrees around 142 and the RPM at 3677, channel 1 shows a lot of missed injections. If I then change the oddfire angle of cylinder 2 to 142 and set the 'injector characteristics' at 0, the issue moves to injector 2.

Hope this helps, the ignitions look just fine in this one so that is great news :)

noisymime, 2024-12-08:
Hmmm OK. I'll try those and see if I can reproduce what you're seeing. Thanks again for the testing.

adbancroft, 2024-12-09:
A couple of shots in the dark for you to try:

1. https://github.com/adbancroft/speeduino/tree/SCHD_7_optimize_FuelSchedule. Uses same basic scheduling algorithm as master, but has a few more checks in it.
2. https://github.com/adbancroft/speeduino/tree/OptimisePw. Uses a fuel scheduling algorithm based more in the time domain than angular (no change to ignition...yet).

YMMV, these haven't been seen by anyone but myself.

noisymime, 2024-12-10:
```
channel2InjDegrees = configPage2.oddfire2;
channel3InjDegrees = configPage2.oddfire3;
channel4InjDegrees = configPage2.oddfire4;
```

This looks like it should be something that is there by default. The oddfire timing currently only changes the ignition angles but it makes sense for them to apply to the injection angles as well. 

> When setting 'injector characteristics' degrees around 142 and the RPM at 3677, channel 1 shows a lot of missed injections. If I then change the oddfire angle of cylinder 2 to 142 and set the 'injector characteristics' at 0, the issue moves to injector 2.

Hmmm even with the change you're mentioned added, I cannot seem to reproduce this at all. I've pushed 1 further small change up on that branch, but I don't think it will have any impact as I can't see this issue with or without it applied. 

I've set it exactly as you've described and tried pulsewidths all the way from 2ms to 17ms and am not seeing issues on any injection channels. I'm testing 4 cylinder, sequential with a 12+1 dual wheel trigger arrangement. This is all on Teensy 4.1, I haven't tried 3.5 but I'd be surprised if it was different. 

I'll keep having a play and see if I can get anything to show up.

noisymime, 2024-12-10:
Just FYI, I've created #1271 to add oddfire angles to the injection timing.

noisymime, 2024-12-11:
OK, I've pushed up a small change to the schedule rework branch that should fix this. 

From my testing, there is roughly a 2 RPM 'window' for any given pulsewidth where this could theoretically occur, so unlikely but not impossible. With this last addition though I can no longer reproduce it under any conditions.

noisymime, 2024-12-11:
May have spoken too soon. New fix coming

noisymime, 2024-12-11:
OK, so turns out this wasn't a problem with the value being 0, it's when the injection start angle value is rapidly changing back and forth between 0 and `CRANK_ANGLE_MAX_INJ`. 

This should be fixed now in the above mentioned branch

RempageR1, 2024-12-11:
> OK, so turns out this wasn't a problem with the value being 0, it's when the injection start angle value is rapidly changing back and forth between 0 and `CRANK_ANGLE_MAX_INJ`.
> 
> This should be fixed now in the above mentioned branch

Yes! Indeed I tested it and it looks good now. Thank you for working in this issue. Great to have it resolved!

RempageR1, 2024-12-11:
> Since we're essentially juggling [modular arithmetic](https://en.wikipedia.org/wiki/Modular_arithmetic), we should limit injection angles to [0, CRANK_ANGLE_MAX_INJ**-1**], aka [0, CRANK_ANGLE_MAX_INJ). E.g. 0-719°. Same for ignition angles.
> 
> That's what https://github.com/adbancroft/speeduino/tree/OptimisePw (mentioned above) does - not sure if it got tested by @RempageR1.

I suggest we wait to see if @noisymime finds this needs to be part of the main Speeduino code. If he does, I`d be happy to do some testing.

noisymime, 2024-12-11:
> Since we're essentially juggling [modular arithmetic](https://en.wikipedia.org/wiki/Modular_arithmetic), we should limit injection angles to [0, CRANK_ANGLE_MAX_INJ**-1**], aka [0, [...]

I'll start having a look at these next week. To be honest this was a bigger problem to work through than I had expected, the deeper I dug the more things I found and it was quite time heavy to test. I'll clean up the branch as it is now and merge it and then start looking at larger changes to the whole scheduling system. 

One thing I've (re)learned coming out of this is that there are a LOT of edge cases that need to be handled and that aren't always obvious or easy to test for. Need to have a think about how best to automate those as they were very time intensive to reproduce. 

@RempageR1 Thanks for all your work in testing this!

commit 97b72b02 (2024-04-05): Include current inj angle in injector timeout calculation. Fixes #1181
commit a94b1231 (2024-04-05): Revert "Include current inj angle in injector timeout calculation. Fixes #1181"

This reverts commit 97b72b027e457bd3303af85f261bfb961b66fbd7.
PR #1272 (2024-12-19): Fuel schedule improvements
This addresses concerns raised in #1181 as well as fixing a number of additional edge case issues with fuel schedulers. 

Fixes #1181
closed 2024-12-19 by noisymime