speeduino/speeduino #1184: Min/max allowable voltage value for O2 feedback.

Admanirv123, 2024-02-21:
Add function to configure min and max voltage for wideband controller to account for controller/sensor failure. 

<= min value sensor/controller deemed malfunctioned. 

>= max value sensor/controller deemed malfunctioned.

E.g less 0.1 deemed malfunctioned. 

Optionally if either condition met deem there is an issue and not use wideband feedback.

jonbills, 2024-02-21:
What would this add that the AFR max and min don't already cover?

Admanirv123, 2024-02-21:
I presume by min and max you mean in calibrate afr table? 

I think even if you set only correct values low and high, it'll still interpolate and think anything for example below 0.5 it will think is a valid afr value and similarly at the higher end of the scale. If we take aem x guage people will put it 0.5 min / 4. 5 max but it'll still use values and presume lower/higher afr below this value unless it simply ignores and doesn't use it for fueling? - maybe it's simply not coded like this and presume there's an error?.

jonbills, 2024-02-22:
No, not in calibration but in tuning->afro2 
https://wiki.speeduino.com/en/configuration/O2
It doesn't apply ego correction if the afr reading in lower or higher than the specified values.

noisymime, 2024-03-15:
As mentioned by @jonbills , this feature is already present in the closed loop O2 control