speeduino/speeduino #1204: Secondary tables switching option for lesser instead of greater

joeclone, 2024-04-07:
Possibility need for conditional switching for secondary fuel and spark tables with an option for "lesser/smaller than" instead of the existing "greater than" for TPS and MAP switching.

jonbills, 2024-04-07:
Just put the values for when tps/map is lower than the switch value in table1

MykkH, 2024-04-15:
In the firmware, secondaryTables.cpp, under the variables in the VE2 strings change your desired condition (or all) from > to <, save and load your modifed firmware then open the .ini file in an editor and change line 2696 under the Table2Dialog_switch to say "is less than" save and load your ini to the project properties. 

I do this to run an idle VE table.

HWright9, 2024-09-03:
Or use the user defined outputs (you can chain them up also) to switch an otherwise un-assigned arduino pin. Then use that same pin as the input to the table switching. Gives you a lot more options for switching, time delays etc.