speeduino/speeduino #1210: Taper on load axis on acc enrich

runesmdk, 2024-04-29:
Like the rpm-taper, is it possible to add a load-axis taper (map or tps)? Driving with throttlebodies and having the injectors somewhat away from the cylinderhead makes the fuel behave not nice during transients with vacuum, needing quite a bit of acc enrich at low to mid tps transient. Going from mid to high tps gives same enrich, but since "manifold pressure" doesnt change at that tps range, it overfuels and bogs. It is nearly impossible to smooth out without loosing either top or bottom drivability. 

So in short; 
Can a load taper be added, just like rpm taper? Or is it possible for a 4x4 table to do a acc.enrich multiplier lookup, with x as rpm and y as load/tps? Like 100% acc enrich at 1000rpm 0% tps(start point), and 25% acc enrich at 4000rpm 50% tps(start point) ?

theKraid, 2024-05-01:
I have made the same obserservations on my ITB setup
Requested it via #697
Unfortunately nothing ever came of it yet

Marlo0oty, 2025-03-17:
I have the same issue on itb on my speeduino. Ive been tuning the ae for two years couldn't get it right. On motec it was very easy as it had an 3d table for fueling and another table which had like a sensitivity for rpms. Some engines have hard timing spinning the first 1,000 rpm then it will take off. What i mean is the taper isn't enough but a 2d table which has rpm/enrichment  "percentage of the actual amount". Would do it all for us.