speeduino/speeduino #1213: 202402 firmware VVT1 no reading and no duty aplied to solenoid.

h25r, 2024-05-17:
Hi!
Im running efi core4 from pazi88 github with no issues on firmware 202305
i updated to latest firmware and im having some problems
VVT1 sensor dont read any value, and no duty aplied to solenoid, closed or open loop dont make any difference.
Also having problems reading CAN inputs and PWM fan output but maybe are caused by the changes on secondary serial.

[CurrentTune.zip](https://github.com/noisymime/speeduino/files/15358395/CurrentTune.zip)

dougsgh, 2024-05-26:
I'm having the same problem. First time build. I can't get any vvt1 or 2 output happening in open or closed loop. Tried multiple settings. I'll try older firmware.

noisymime, 2024-05-26:
Are you able to provide a log in addition to your tune?

noisymime, 2024-05-26:
This looks to have been a change in the way the secondary tooth is handled when the `Poll` option is selected. 

The Poll option was never specifically intended to be usable with closed loop VVT, but I can reinstate the same behaviour as was there previously. 

This doesn't impact any of the other secondary VVT patterns

noisymime, 2024-05-26:
This should be resolved now. 

If you're able to try the Master firmware version (Available in SpeedyLoader at the bottom of the list) and confirm, that would be good.

closed 2024-05-26 by noisymime

dougsgh, 2024-05-27:
Thanks noisymime!

Should I be downloading a new ini file from somewhere? I clicked connect anyway.

So with the latest 'master' firmware loaded in, I am now able to see cam angles on both cams, however I am still not able to get any vvt1 or 2 output happening in open or closed loop. On startup, no cam angle is shown until after a composite log. The cam angle does not appear to change when manually actuating the VVT solenoid (which almost stalls the engine - so is definitely changing)

A few project details and some plot thickeners...
This is for a Y62 patrol with the later model v8 in the earlier model, series 1. The S1 has vvt on inlet cams only, the later model engine, which I now have running here (Perth, WA) has vvt on exhaust cams too. The original ecu is running everything apart from the exhaust VVT. I have a 0.4 board running piggyback off the stock crank sensor, and independently connected to both exhaust cam position and vvt solenoids.  I dont have MAP or coolant temp connected as yet which might upset things.
The other thing that might upset things is the stock triggers arent supported, crank is 36-2-2, 180 degrees apart, you will notice I have the tune set to 18-2 (seems to work and sync, obviously rpm is double, but not serious for VVT only). The other non supported piece is the stock cams are 4T. Running the crank as half a crank, it would only be a 2T, although, I realise still not supported. (I was hoping to get running and then explore some firmware tweaks).

This is becoming a common engine swap in Australia due age of S1 and availability/cost of s1 engines vs s5 engines. The majority are going down the route of s1 fixed exhaust cam sprocket swap, but I felt the need for some discovery.

In essence, I realise there are probably better guinnea pigs out there, and this should be a new feature request with a healthy donation to the cause which I am happy to do!
[VK56 VVT.zip](https://github.com/noisymime/speeduino/files/15453510/VK56.VVT.zip)

dougsgh, 2024-05-27:
[image]

[image]

[image]

h25r, 2024-05-27:
> This should be resolved now.
> 
> If you're able to try the Master firmware version (Available in SpeedyLoader at the bottom of the list) and confirm, that would be good.
[TestMaster.zip](https://github.com/noisymime/speeduino/files/15461913/TestMaster.zip)

Not working for me.
Add log and tune.

noisymime, 2024-05-28:
> Thanks #noisymime
>
> Should I be downloading a new ini file from somewhere? I clicked connect anyway.
>
> So with the latest 'master' firmware loaded in, I am now able to see cam angles on both cams, [...]

OK, that (and your pictures) clears things up a bit. 

First off that trigger arrangement is never going to work with the `Poll` secondary type. Poll is designed for use with things like a 'half moon' cam pattern where the level of the cam reading changes at tooth #1 

In your case I'd say you should be using the 'Single tooth cam' option, however that 4T tooth on the cam isn't going to work for closed loop VVT either. 

Ultimately I think it's going to need a proper decoder for this setup as trying to band-aid it with patterns that don't really fit isn't ideal.

I'll see if I can dig up any proper timing charts for this pattern.

dougsgh, 2024-05-28:
> OK, that (and your pictures) clears things up a bit.
>
> First off that trigger arrangement is never going to work with the `Poll` secondary type. Poll is designed for use with things like a 'half [...]

Appreciated, thanks!

I am a little confused however, the trigger setting I have is set to single tooth, am I missing something?

[image]

[image]

In regards to decoders, one stumbling block for this engine is sync must ultimately be determined by the stock ecu using the intake cam 1 and 2 (not connected to speeduino here) patterns which are 4+1+1+0+0 and 4+0+0+1+1 for the respective banks. The crank being 36-2-2 (16 T - 2, 16 - 2) and the 4T exhaust cams arent going to determine "sync". Given that I dont need fuel or spark, I was hoping to get away without achieving sync. If i can avoid cutting 3 teeth off each cam, that would be ideal as by opening the new factory crate engine, I'd void the warranty.

dougsgh, 2024-05-28:
Ok, issue resolved. I figured out that vvt wasn't happy without a valid coolant temperature. 

I first rolled back to 23 firmware, that didn't work, once I added coolant temp sensor, things came to life, rolled forward to 24 release current firmware, and vvt1 and 2 outs are working in open loop, closed loop not working quite right due to my 4t tooth setup. It does log a fixed incorrect cam angle after a tooth log and I was able to get PID to try and chase it and put some pwm out. 

Apologies for the newbie errors here!

Now for a crash course in C++ and decoder logic 🤣

h25r, 2024-05-28:
> > Not working for me. Added log and tune. [TestMaster.zip](https://github.com/noisymime/speeduino/files/15461913/TestMaster.zip)
> > If i start a composite log, start reading vvt1 angle but in error state
> 
> What is your actual trigger arrangement on the crank/cam?
> 
> Your tune is set to use the `Poll` secondary as well, do you have a half-moon cam trigger?

The crankshaft has 60 teeth with 2 missing, and the camshafts are half-moon.
Motor is m52tub25 dual vanos. both vanos have half-moon trigger

noisymime, 2024-05-29:
> I am a little confused however, the trigger setting I have is set to single tooth, am I missing something?

Sorry, I had mixed up to the 2 tunes in this issue and was looking at the one from @h25r!

Unfortunately I do think that a dedicated decoder pattern is going to be your only option. Do you have any timing diagrams for the pattern at all?

noisymime, 2024-05-29:
> > This should be resolved now.
> > If you're able to try the Master firmware version (Available in SpeedyLoader at the bottom of the list) and confirm, that would be good.
> 
> Not working for me. Added log and tune. [TestMaster.zip](https://github.com/noisymime/speeduino/files/15461913/TestMaster.zip)
> 
> If i start a composite log, start reading vvt1 angle but in error state
> 
> [2024-05-27_23.43.41test2.zip](https://github.com/noisymime/speeduino/files/15461954/2024-05-27_23.43.41test2.zip)

OK, spun this up to test today. What I can see in your log is that the VVT1 angle reading is exceeding your Max Cam angle setting of 42. This is what is putting the VV1 into an error state and turning off the output as a failsafe. 

See what happens if you increase your Max Cam Angle setting up to 43 or 44. That should enable the output and clear the error that is showing. 
On the bench here I took your tune, upped the Max Cam Angle to 50 and am getting a VVT output as expected.

noisymime, 2024-05-29:
@h25r Actually, I've also just pushed up 1 more small change that might help in your specific case. Try the master firmware again from SpeedyLoader and see if you get any difference.

dougsgh, 2024-05-29:
> Sorry, I had mixed up to the 2 tunes in this issue and was looking at the one from @h25r!
>
> Unfortunately I do think that a dedicated decoder pattern is going to be your only option. Do you have any [...]

Thank-you, that would be amazing! 

I mapped out the pattern in excel, cam in locked position. The cam tooth aligns 8 teeth after the gap in the crank trigger for both cams. (vvt1&2) (This can be seen in the scope trace picture above). 

Unfortunately no way to get sync without the inlet cam's, however I have no plans in running fuel or spark. I only plan for this to be a VVT controller.

[VK56TriggerExhaust.xlsx](https://github.com/noisymime/speeduino/files/15478837/VK56TriggerExhaust.xlsx)

h25r, 2024-05-29:
> @h25r Actually, I've also just pushed up 1 more small change that might help in your specific case if you're getting 0 for VVT readings. Try the master firmware again from SpeedyLoader and see if you get any difference.

working perfect now with the change you suggested. Thank you very much, i will sent couple beers to u!

dougsgh, 2024-05-31:
[image]

I was hoping this was just a tunerstudio display issue, but i confirmed that I do not have vvt1 output until i plug in ts and hit tooth log. Those PID settings are doing great though!

noisymime, 2024-05-31:
@dougsgh Which decoder are you using now, the missing tooth one? If so, try the master firmware and see if it's any change.

Also.... I'm guessing the RPM figure is doubled? 🤣

dougsgh, 2024-06-01:
@noisymime Winning! Thank-you, master works! 2 stroke 5.6l v8, revving to 12000rpm would be a beast 🤣. Running missing tooth, and single tooth cam. No change to the stock 36-2-2 crank or 4t cams. Those pids have the cam angle following incredibly well. Bit more street tuning and then some dyno time. 👍

commit bff1beca (2024-05-26): Record VVT angle when Poll cam method is in use. Fixes #1213
commit 68be5c03 (2024-05-26): Record VVT angle when Poll cam method is in use. Fixes #1213