speeduino/speeduino #1214: Stepper OL+CL control

paoloponzano, 2024-05-20:
Hi! This is not an issue report but rather an ask for help understanding how the code works. I'm currently running a speeduino UA4C (V3B) on a first gen fiat panda 1000cc that I converted from carburetor to MPI. 
I've been having some trouble having the idle control stepper motor to behave correctly but I now reached a stable setup with just OL. My open loop table goes from ~95 steps cold to ~45 when warm, cranking is 110-80. It seems to loose steps if moved too much and, not having a failsafe spring or something, it would eventually go completely open. I already replaced the actuator but I'll also have a closer look at the stepper controller in the near future, it might not be delivering enough current. 
With this being said, I've also been experimenting with closed loop idle control, especially OL+CL. If i understood correctly, the OL table is fed to the pid algorithm as reference point to start calculations. And when pressing the accelerator or revving past 'idle target rpm + integral rpm reset' only the integrative function is disabled and reset to 0. There are a couple of issues with this behavior in my case. When going off idle, the integrative is reset but proportional and derivative still fight against my accelerator input closing the stepper until 'min steps'. And, since i need long steps time, the actuator isn't fast enough to open when going back to idle. 
Since the settings for closed loop take in absolute steps values 'max steps, min steps', i need to consider both cold and warm operation. This means that i need to set max steps to at least 95 and min to less than 45 which gives too much room for the pid to move the stepper. If engine is warm, and i try to move the car just releasing the clutch and no accelerator (for example parking) the control will try to open the stepper up to max steps (which is a lot when warm) and might even eventually loose steps and that's not good. I know i must fix hardware problems first but this might help others too.
What I'd like to suggest is to make the OC+CL code use the 'max steps' and 'min steps' as relative values instead of absolute. That means how many steps the closed loop control can add or subtract to the open loop table. For example when, setting max 10 and min 10 would allow pid to move the stepper from 85 to 105, while when warm it could go from 35 to 55. This wouldn't allow for wild oscillations even if the pid parameters are wrong and we could stop the whole pid correction when tps> set amount so that the stepper maintains the last position when going off idle. 
I tried editing the idle.cpp file but I couldn't make it work, i don't really understand how the bit shifting works so your help would be appreciated 🙂
Pure CL control should not be modified ofc
Here is my draft https://github.com/paoloponzano/speeduino

Thank you and have a great day c:

paoloponzano, 2024-05-20:
Oh i forgot to mention I'm also using idle advance control, so that's why it's still ok if the stepper moves slowly

VitorBoss, 2024-05-20:
Hi, the control act very fast, if is a delay might be the step duration been too long, usually 3 or 4ms is enough.
About the control fighting you, the PID ninbers are 1.0 for 100% correction, keep them low, also try to use low P as it apply over RPM difference.
I use it on my engine, PID here are 0.09, 0.25, 0.055 and 0.5% for integral reset. Integral reset is the key here, when is abobe this value the controller acts in open loop, when you set this too high it fights against you!

paoloponzano, 2024-05-20:
Hi @VitorBoss ! Thank you for the reply
I've tried setting the parameters starting from 0 and increasing in small steps as you say, and integral reset set to 1%. To me it seemed like even when above that tps value, the proportional part would close the valve instead of going back to OL steps... 
Today i tried setting only P=0.03, I=0.03, D=0 and it seems to operate slowly enough to act as a 'long term' correction while not loosing steps. But it would work even better if this correction could be kept instead of moving back to OL steps. And i might be able to use more aggressive settings if closed loop was more constrained as i explained. 
I'm not saying the way the code is needs to be changed for everyone, I'd just like to give it a try on my setup!

VitorBoss, 2024-05-20:
If that happens only cold can be humidity inside it, the seals don't let dirty enter the sensor, but if they are wet or the engine bay was washed with high pressure water jet, water could pass the seals and behave finicky when cold but normal with hot engine.
Just a guess though.

paoloponzano, 2024-06-02:
> Hi, the control act very fast, if is a delay might be the step duration been too long, usually 3 or 4ms is enough.
>
> About the control fighting you, the PID ninbers are 1.0 for 100% correction, keep [...]

So after some fiddling I was able to get good idle control with stock code without loosing steps 👌🏻👌🏻
But looking at a log, it seems like the closed loop still tries to move the valve a lot instead of keeping OL steps when tps< "integral reset tps" and rpm> "hysteresis rpm".
Shouldn't it keep the valve at OL steps when one of these variables is true? (tps higher than 0.5 or rpm higher than idle rpm target + 100)
Here's a pic to understand better
For reference, at that temperature the OL steps are 66, integral reset tps is 0.5% and integral reset rpm hysteresis is 100
[image]
[image]
[image]