speeduino/speeduino #1217: CAN RusEfi WBO not working together with BMW or VAG CAN broadcasting

Zazins, 2024-06-07:
If RusEFI Can option is enabled, BMW and VAG broadcast stops working, 
Should be possible to use them together. 
Thanks!

VitorBoss, 2024-06-07:
Do you have such controller installed?

Zazins, 2024-06-07:
Yes I have, but for tests you dont need RusEfi CAN  Wbo. Just set option to enable and dash broadcast will stop woking after ecu reset.

VitorBoss, 2024-06-07:
You can't just enable it if you don't have it connected, speeduino send a message to it and if there is no device to receive the CAN module will enter in error state and stop sending messages after 127 consecutive fails. That is a feature of CAN bus, you can't send a message without a listening device, there are handshakes involved in the CAN bus.

Zazins, 2024-06-08:
Yes, i underatand that, but this is not the case. When sensor is connected and option is set to enable for RusEfi. Can dash broadcast stops working.  There should be something else. 
Im 100% sure that sensor and Can messages from RusEfi are ok. This should be at Speeduino side.

Zazins, 2024-06-08:
It seens that the the RusEfi wbo functiona uses can0.write as well as dashmesaage use it. Without a queue functions in code it ignores all other messages to be sent.  Easy fix is just add in void recievecanwbo to keep sending dashmessages for bmw or vag. If i look in code it seems it will happen if using can_commans etc.

VitorBoss, 2024-06-09:
That isn't the problem, the CAN library queues it

Zazins, 2024-06-10:
Yes. It does. Looked in the code. 
I have a fix for this, but the root cause is somewhere else.  As the RusEfi Wbo only impact this.

Zazins, 2024-06-11:
Root cause of this issue is that the option RusEfi WBO sets CAN ID to extended, putting extended outMsg.flags.extended = 1;
All other STD CAN id cant go true.  So the fix for this is to add for all outgoing can id flags outMsg.flags.extended = 0; or outMsg.flags.extended = 1;. This will fix all problems. 

If someone has or had issues could update code in comms_CAN.cpp
And maybe @noisymime could pull in main. 

Thanks!

VitorBoss, 2024-06-11:
Nice catch, what if instead of inserting `outMsg.flags.extended = 0;` on all messages we add it after the rusEFI message is [sent](https://github.com/noisymime/speeduino/blob/master/speeduino/comms_CAN.cpp#L87)?
This would be a very minimal and simple fix, could you test it?

Zazins, 2024-06-11:
It will work as it should, I was testing this approach before. I just thinking how about adding extended CAN id's later? Someone will forget about setting it back and will result in same situation. I'm not sure what would be better approach for Speeduino.

noisymime, 2024-07-30:
Believe this was resolved with #1223 ?

closed 2024-07-30 by noisymime