speeduino/speeduino #1218: Incorrect BMW Can broadcast Coolant value conversation.

Zazins, 2024-06-07:
Incorrect Can message conversation leads to incorrect coolant temperature displayed.
Incorrect value is more then half wrong from correct vales. 

Correct conversation with very low incorrect value keeping old conversation style.

noisymime, 2024-07-30:
@pazi88 Are you able to confirm here at all? 

It does appear that the `CALIBRATION_TEMPERATURE_OFFSET` value in `comms_CAN.cpp` is currently being subtracted when that may not be needed. 

From what I can find online the conversion should be:
`Temp in C = .75 * byte - 48.373`
or to rework it for the byte value:
`byte = (C + 48.373)/.75` 
aka:
`temp_CLT = (currentStatus.coolant + 48)*4/3; //CLT conversion`

I don't have a BMW setup to test this on. @Zazins Are you able to test this with the above line?

Zazins, 2024-07-30:
Hi! 

This is correct calculation that Im using on other projects Calculation = (HEX * 0.75) - 48°C, but I dont know why this calculation leads to incorrect values by Speeduino. If i sent coolant temperature over Serial and then calculate temperature then calculation is correct. 
However yes I can test it.

noisymime, 2024-07-30:
> (HEX * 0.75) - 48°C, but I dont know why this calculation leads to incorrect values by Speeduino.

That calculation would be to turn whatever is sent on CAN back into degrees C. What Speeduino needs to do is the opposite of this, convert degrees C into the value that is expected on the bus. 

I believe this line should be correct:

`temp_CLT = (currentStatus.coolant + 48)*4/3; //CLT conversion`

pazi88, 2024-07-31:
> @pazi88 Are you able to confirm here at all?
>
> It does appear that the `CALIBRATION_TEMPERATURE_OFFSET` value in `comms_CAN.cpp` is currently being subtracted when that may not be needed.
>
> From what [...]

I haven't paid close attention to the Speeduino code lately. But isn't the  `currentStatus.coolant` still offset by 40 degrees? Meaning that 0 raw value on the `currentStatus.coolant` is actually -40°C. So that's why the `CALIBRATION_TEMPERATURE_OFFSET` is there.

Let's think of the 0°C for example. That's 40 in raw value in speeduino. So to get the actual temp, we need to subtract -40 from that. Then to convert that to the BMW value, we need to add the 48 offset there and then reverse the 3/4. So in my head the current calculation is perfectly correct.

I have run the speeduino years in my BMW with the CAN cluster and in that the temp has been correct. But I didn't actually run the STM32, but I did read the data from serial3 and did the conversion separately. In that I have the conversion like this: 'CLT = (currentStatus.CLT -40)*4/3+64;'

It's years since I did this, but there's probaply reason why I moved the offset by 48 after the multiply/division and include multiply/division in the offset value (48*4/3 = 64).

pazi88, 2024-07-31:
> That calculation would be to turn whatever is sent on CAN back into degrees C. What Speeduino needs to do is the opposite of this, convert degrees C into the value that is expected on the bus.
>
> I [...]

umm, yes. You have the parentheses there. That the current calculation doesn't. So I think that does the same thing as I did with my own calculation in the previous message. But that conversion is still from the RAW value, not the actual temp from °C. So it would be `temp_CLT = ((currentStatus.coolant - CALIBRATION_TEMPERATURE_OFFSET + 48)*4)/3;`. but because the `CALIBRATION_TEMPERATURE_OFFSET` is 40, in short it's: `temp_CLT = ((currentStatus.coolant + 8)*4)/3;`

noisymime, 2024-07-31:
> But isn't the currentStatus.coolant still offset by 40 degrees? Meaning that 0 raw value on the currentStatus.coolant is actually -40°C. 

No, though it's a very easy mistake to make (I had to double check this myself). The `CALIBRATION_TEMPERATURE_OFFSET` amount is what the calibration table is offset by and how much the value is offset by when communicating with TS, but this is taken into account in the `readCLT()` function so `currentStatus.coolant` is a true signed value. If the coolant is -10, then `currentStatus.coolant` will be -10 and not 30. 

>  You have the parentheses there.

My brain likes to have these things as clear as possible rather than relying on the compiler not try and do any sneaky optimisations that may impact the integer division accuracy 🤣
The parenthesis shouldn't be changing the intent of the calculation (Eg dividing the offset coolant value by 0.75), but it does help make it very clear what order things should be happening in.

pazi88, 2024-07-31:
> No, though it's a very easy mistake to make (I had to double check this myself). The `CALIBRATION_TEMPERATURE_OFFSET` amount is what the calibration table is offset by and how much the value is [...]

You are right. The offset seems to be added for secondary serial, so that's why I need that in my other thingy. 

So the `temp_CLT = ((currentStatus.coolant + 48)*4)/3; //CLT conversion` should be correct.

I'm just thinking that maybe the `temp_CLT` should be `int16`, instead of `uint16` for the calculation to work correctly. When the `currentStatus.coolant` is also signed.

adbancroft, 2024-07-31:
FYI, in one of my local branches (soon to be a PR) I have the beginnings of a zero-footprint formal system of converting between "tune" values and "working" values. It makes these type of bugs very clear - I noticed this issue (and a few others) but haven't had time to raise them (apologies).

Ideally we would use something like [boost::units](https://www.boost.org/doc/libs/1_85_0/doc/html/boost_units.html), but that requires a whole lot of prerequisites.

noisymime, 2024-07-31:
> I'm just thinking that maybe the `temp_CLT` should be `int16`, instead of `uint16` for the calculation to work correctly. When the `currentStatus.coolant` is also signed.

It *shouldn't* matter as it has the 48 added to it, so it will never come out negative. It's then converted to a uint8_t for transmission anyway, so even if it was an int16_t it would be converted to unsigned when put into the message buffer.

noisymime, 2024-09-20:
Has anyone had the change to test this proposed change? I'm happy to push it in as is given the confidence, however it would be nice if someone is able to test it first. 

To recap, line 223 in `comms_CAN.cpp` should change from:

`temp_CLT = (((currentStatus.coolant - CALIBRATION_TEMPERATURE_OFFSET) + 48)*4/3);`

to

`temp_CLT = ((currentStatus.coolant + 48)*4)/3;`

commit b99ab35f (2024-11-28): Correct Coolant reading over BMW CAN protocol. Fixes #1218
closed 2024-11-28 by noisymime