speeduino/speeduino #1224: TunerStudio MS changing values randomly

Nikolai521, 2024-06-29:
Well i dont know if report this here, but when i open TS with the ecu disconnected, it always changes something, most of the time changes the stepper value to impossible ones, like this one

[image]
The mines are 120 and 30! :skull: 

Also i dont get any error when i open it, if i use the ecu connected i dont get the "changes" window and the values are normal, only on "off-line", also sometimes it writes that to the ecu and the engine shoots to 3000rpm because the idle valve tries to do +300steps

I have everything update, last update on TS and last update on Speeduino (0.4)

noisymime, 2024-07-07:
Is it only specific settings that this always happens with?

Just to double check, you've loaded the ini file version that matches your tune?

Nikolai521, 2024-07-08:
I noticed, Actually is just the stepper values and the AFR table generator are totally broken, just fills the table with 1 and 0
[image]

Also yes, i uploaded again the firmware and downloaded the new ini, but still the same, but this happends ONLY if i start TS without the board connected, if i plug the board the changes gets written without any warning

Actually i dont know about this issue before, i just started to use the stepper closed loop these days and i noticed that, just i need to connect the board first and TS in second, but this is an issue when im using bluetooth and it takes a bit more longer (engine skyrocketing again to 3000rpm lol)
Also the AFR generator worked well 3 months later when i generated it

if you want to try, i leave my tune, i dont see another one with this issue, even on my 2 computers i have the same issue (both different operating systems, but still the same thing)

[test.zip](https://github.com/user-attachments/files/16121555/test.zip)

noisymime, 2024-07-08:
What version of TS are you using?

Nikolai521, 2024-07-08:
3.2.03

ponzano, 2024-09-01:
Happens to me too, exactly the same 
Has been doing this for about 7 months (when i started using idle closed loop) but might have started sooner

Wakamoly, 2024-09-02:
I'll add to this as well, my experience is similar on Linux desktop and Raspian as well as Windows 10. I thought this was specific to me with the amount of computers I change between, which could be a contributor though.

Nikolai521, 2026-01-11:
> Sounds like this a question to send to [support@efianalytics.com](mailto:support@efianalytics.com)

I already did, plus some friends and never got response, only happens on speeduino sadly :c

Nikolai521, 2026-01-11:
> Can you attach a comms debug log?

Today i only have the debug log from TS, no longer using speeduino sadly :c 

[TunerStudioDebugPackage(1).zip](https://github.com/user-attachments/files/24555048/TunerStudioDebugPackage.1.zip)

8SecSleeper, 2026-10-31:
This happens, tuner studio on load is reading the tune and doing the math based on it being PWM selected, but then it sees stepper is selected in the tune read from the ecu, then TS applies a 3x multiplication. 

I've been sitting here assuming I caused the bug myself, then decided to check issues on here and the bug has existed for years now and never resolved.

Nikolai521, 2026-11-02:
>          I've been sitting here assuming I caused the bug myself, then decided to check issues on here and the bug has existed for years now and never resolved.

God i remember when i sent a mail to EFI Analytics blaming TS, obviusly got no answer