speeduino/speeduino #1226: Fool-proof correction tables and other improvements

roko772, 2024-07-12:
Hi, I'm not Speeduino user, but as an mechanical engineer and some sort of motorsport enthusiast with lower-than-basic programming skills, I decided to look at the code and TunerStudio settings and I've some conclusions. I hope, it will improve Speeduino by making it more fool-proof and easier to use.

**1. IAT, barometric and fuel temperature/pressure correction**
I've downloaded some base tune for BMW M43 and here I saw nothing more than plain 2D table with correction table...
[image]
despite being fully replaceable by ideal gas law. In fact, you don't need nothing more than ratio between 20°C and IAT (both in Kelvin). See any difference? I don't ;)
[image]
If you really need to implenent adjustable IAT correction, just do it in -127%...127% range with 0% as baseline. The same thing applies to barometric correction.
Speaking of fuel temperature/pressure correction, I think it should be available all the time, not only when Flex Fuel is on. And again, try to use math in the first place.

**2. Warmup Enrichment**
Why Speeduino allows us to set WUE multiplier below 100%? Everyone knows that it must not be lower than 100% under no circumstances, no matter what, NEVER EVER EVER NEVER, you name it.

**3. VE as ignition load source?**
I got access to free Russian software for their domestic ECU tuning. They use air consumption per cycle measured in mg/cycle as engine load factor. This number can be roughly reffered to the volumetric efficiency, so it will be nice to see VE as load factor for ignition advance map. RusEFI can use "% cyl filling" for instance.

**4. CLT dependent rev-limit?**
I don't think it needs any explaination ;)

**5. Idle/EGO PID autotune?**
There are some existing algorithms for PID automatic tuning, let's try to use them for idle or EGO.

**6. Custom Arduino Core? (compatibility, performance)**
I saw some optimized Arduino cores for ATmega328, for example GyverCore where its authors stated about serious performance improvement over standard Arduino core. Maybe it will run better with optimized core?

pazi88, 2024-07-13:
1: this map is actually something I tune very often. We don't live in perfect world, so IAT correction doesn't behave based on the ideal gas law. There is error from the temp sensors etc. So it pretty luch needs to stay as tunable map.

4: already there.

roko772, 2024-07-13:
1: Still, I think that "baseline" (whatever we call it) should be calculated by IGL. If it's 5% off from reality you should set 5%, if -15% - you set -15%. First IGL, and only then - correction chart. Defining a "dumb" curve without any pre-calculation is IMHO incorrect.

noisymime, 2024-07-13:
1. As mentioned, the ideal gas law is a good starting point, but it very rarely works perfectly in a real world setup. The default IAT correction in the base tunes though is simply the IGL put into a 2d curve, so you're already getting what you're asking for, but in a way that can be adjusted if required. 

> must not be lower than 100% under no circumstances, no matter what, NEVER EVER EVER NEVER

Except that there are cases where this is useful. Usually its around a tune where the primary fuel table was done for ethanol and warm starting on regular fuel will require less. No this isn't common, but it's better to have the option and not need it than vice versa. 

3. 'VE' is effectively what the value in the fuel table is, not what should be on the axis. Granted it's a loose definition of VE if its a boosted engine, but for NA that table is effectively defining what the VE will be at a given RPM and load (MAP or TPS). Measuring that load in actual air flow is what would typically be done with a MAF sensor, which is not supported (And unlikely to be anytime soon). 

4. Need more explanation because this is already there. 

5. Happy to see an implementation of this if you want to submit a PR. PID autotuning can be a bit of a misnomer as it's often very application specific still, but it's something I'd be happy to add. 

6. Is there any core for a 2560 that is similar? Most of the time critical, slower Arduino calls are already worked around in the Speeduino code anyway, so I'd be surprised if something like this made much of a difference, but is there's one that already exists I'm happy to try it out.

noisymime, 2024-07-30:
Closing. If any of these can be progressed individually in a more concrete form another issue can be opened for them.

closed 2024-07-30 by noisymime