speeduino/speeduino #1232: Issue with stm32f103c8t6 (Bluepill 128k) compiling with TIMERS erros

ArtCodeZen, 2024-07-30:
In file included from speeduino/globals.h:108,
                 from speeduino/acc_mc33810.h:5,
                 from speeduino/acc_mc33810.cpp:1:
speeduino/board_stm32_official.h: In function 'void FUEL5_TIMER_ENABLE()':
speeduino/board_stm32_official.h:269:47: error: 'TIM5' was not declared in this scope; did you mean 'TIM1'?
  269 | static inline void FUEL5_TIMER_ENABLE(void) {(TIM5)->CR1 |= TIM_CR1_CEN; (TIM5)->CR1 |= TIM_CR1_CEN; (TIM5)->SR = ~TIM_FLAG_CC1; (TIM5)->DIER |= TIM_DIER_CC1IE;}
      |                                               ^~~~
      |                                               TIM1
speeduino/board_stm32_official.h: In function 'void FUEL6_TIMER_ENABLE()':
speeduino/board_stm32_official.h:270:47: error: 'TIM5' was not declared in this scope; did you mean 'TIM1'?
  270 | static inline void FUEL6_TIMER_ENABLE(void) {(TIM5)->CR1 |= TIM_CR1_CEN; (TIM5)->CR1 |= TIM_CR1_CEN; (TIM5)->SR = ~TIM_FLAG_CC2; (TIM5)->DIER |= TIM_DIER_CC2IE;}
      |                                               ^~~~
      |                                               TIM1
speeduino/board_stm32_official.h: In function 'void FUEL7_TIMER_ENABLE()':
speeduino/board_stm32_official.h:271:47: error: 'TIM5' was not declared in this scope; did you mean 'TIM1'?
  271 | static inline void FUEL7_TIMER_ENABLE(void) {(TIM5)->CR1 |= TIM_CR1_CEN; (TIM5)->CR1 |= TIM_CR1_CEN; (TIM5)->SR = ~TIM_FLAG_CC3; (TIM5)->DIER |= TIM_DIER_CC3IE;}
      |                                               ^~~~
      |                                               TIM1
speeduino/board_stm32_official.h: In function 'void FUEL8_TIMER_ENABLE()':
speeduino/board_stm32_official.h:272:47: error: 'TIM5' was not declared in this scope; did you mean 'TIM1'?
  272 | static inline void FUEL8_TIMER_ENABLE(void) {(TIM5)->CR1 |= TIM_CR1_CEN; (TIM5)->CR1 |= TIM_CR1_CEN; (TIM5)->SR = ~TIM_FLAG_CC4; (TIM5)->DIER |= TIM_DIER_CC4IE;}
      |                                               ^~~~
      |                                               TIM1
speeduino/board_stm32_official.h: In function 'void FUEL5_TIMER_DISABLE()':
speeduino/board_stm32_official.h:274:48: error: 'TIM5' was not declared in this scope; did you mean 'TIM1'?
  274 | static inline void FUEL5_TIMER_DISABLE(void) {(TIM5)->DIER &= ~TIM_DIER_CC1IE;}
      |                                                ^~~~
      |                                                TIM1
speeduino/board_stm32_official.h: In function 'void FUEL6_TIMER_DISABLE()':
speeduino/board_stm32_official.h:275:48: error: 'TIM5' was not declared in this scope; did you mean 'TIM1'?
  275 | static inline void FUEL6_TIMER_DISABLE(void) {(TIM5)->DIER &= ~TIM_DIER_CC2IE;}
      |                                                ^~~~
      |                                                TIM1
speeduino/board_stm32_official.h: In function 'void FUEL7_TIMER_DISABLE()':
speeduino/board_stm32_official.h:276:48: error: 'TIM5' was not declared in this scope; did you mean 'TIM1'?
  276 | static inline void FUEL7_TIMER_DISABLE(void) {(TIM5)->DIER &= ~TIM_DIER_CC3IE;}
      |                                                ^~~~
      |                                                TIM1
speeduino/board_stm32_official.h: In function 'void FUEL8_TIMER_DISABLE()':
speeduino/board_stm32_official.h:277:48: error: 'TIM5' was not declared in this scope; did you mean 'TIM1'?
  277 | static inline void FUEL8_TIMER_DISABLE(void) {(TIM5)->DIER &= ~TIM_DIER_CC4IE;}
      |                                                ^~~~
      |                                                TIM1
*** [.pio/build/bluepill_f103c8/src/acc_mc33810.cpp.o] Error 1
In file included from speeduino/globals.h:108,
                 from speeduino/SD_logger.cpp:1:
speeduino/board_stm32_official.h: In function 'void FUEL5_TIMER_ENABLE()':
speeduino/board_stm32_official.h:269:47: error: 'TIM5' was not declared in this scope; did you mean 'TIM1'?
  269 | static inline void FUEL5_TIMER_ENABLE(void) {(TIM5)->CR1 |= TIM_CR1_CEN; (TIM5)->CR1 |= TIM_CR1_CEN; (TIM5)->SR = ~TIM_FLAG_CC1; (TIM5)->DIER |= TIM_DIER_CC1IE;}
      |                                               ^~~~
      |                                               TIM1
speeduino/board_stm32_official.h: In function 'void FUEL6_TIMER_ENABLE()':
speeduino/board_stm32_official.h:270:47: error: 'TIM5' was not declared in this scope; did you mean 'TIM1'?
  270 | static inline void FUEL6_TIMER_ENABLE(void) {(TIM5)->CR1 |= TIM_CR1_CEN; (TIM5)->CR1 |= TIM_CR1_CEN; (TIM5)->SR = ~TIM_FLAG_CC2; (TIM5)->DIER |= TIM_DIER_CC2IE;}
      |                                               ^~~~
      |                                               TIM1
speeduino/board_stm32_official.h: In function 'void FUEL7_TIMER_ENABLE()':
speeduino/board_stm32_official.h:271:47: error: 'TIM5' was not declared in this scope; did you mean 'TIM1'?
  271 | static inline void FUEL7_TIMER_ENABLE(void) {(TIM5)->CR1 |= TIM_CR1_CEN; (TIM5)->CR1 |= TIM_CR1_CEN; (TIM5)->SR = ~TIM_FLAG_CC3; (TIM5)->DIER |= TIM_DIER_CC3IE;}
      |                                               ^~~~
      |                                               TIM1
speeduino/board_stm32_official.h: In function 'void FUEL8_TIMER_ENABLE()':
speeduino/board_stm32_official.h:272:47: error: 'TIM5' was not declared in this scope; did you mean 'TIM1'?
  272 | static inline void FUEL8_TIMER_ENABLE(void) {(TIM5)->CR1 |= TIM_CR1_CEN; (TIM5)->CR1 |= TIM_CR1_CEN; (TIM5)->SR = ~TIM_FLAG_CC4; (TIM5)->DIER |= TIM_DIER_CC4IE;}
      |                                               ^~~~
      |                                               TIM1
speeduino/board_stm32_official.h: In function 'void FUEL5_TIMER_DISABLE()':
speeduino/board_stm32_official.h:274:48: error: 'TIM5' was not declared in this scope; did you mean 'TIM1'?
  274 | static inline void FUEL5_TIMER_DISABLE(void) {(TIM5)->DIER &= ~TIM_DIER_CC1IE;}
      |                                                ^~~~
[... 41 lines cut ...]
      |                                                ^~~~
      |                                                TIM1
speeduino/board_stm32_official.h: In function 'void FUEL6_TIMER_DISABLE()':
speeduino/board_stm32_official.h:275:48: error: 'TIM5' was not declared in this scope; did you mean 'TIM1'?
  275 | static inline void FUEL6_TIMER_DISABLE(void) {(TIM5)->DIER &= ~TIM_DIER_CC2IE;}
      |                                                ^~~~
      |                                                TIM1
speeduino/board_stm32_official.h: In function 'void FUEL7_TIMER_DISABLE()':
speeduino/board_stm32_official.h:276:48: error: 'TIM5' was not declared in this scope; did you mean 'TIM1'?
  276 | static inline void FUEL7_TIMER_DISABLE(void) {(TIM5)->DIER &= ~TIM_DIER_CC3IE;}
      |                                                ^~~~
      |                                                TIM1
speeduino/board_stm32_official.h: In function 'void FUEL8_TIMER_DISABLE()':
speeduino/board_stm32_official.h:277:48: error: 'TIM5' was not declared in this scope; did you mean 'TIM1'?
  277 | static inline void FUEL8_TIMER_DISABLE(void) {(TIM5)->DIER &= ~TIM_DIER_CC4IE;}
      |                                                ^~~~
      |                                                TIM1
*** [.pio/build/bluepill_f103c8/src/SD_logger.cpp.o] Error 1
*** [.pio/build/bluepill_f103c8/src/TS_CommandButtonHandler.cpp.o] Error 1
=============================== [FAILED] Took 18.45 seconds ===============================

Environment      Status    Duration
---------------  --------  ------------
bluepill_f103c8  FAILED    00:00:18.451
========================== 1 failed, 0 succeeded in 00:00:18.451 ==========================

ArtCodeZen, 2024-07-30:
With stm32f4 blackpill compile without problems, just for the stm32f1 bluepill, there are problems with timer 4 and 5.

VitorBoss, 2024-07-30:
BluePill isn't supported anymore, the support was dropped in 2020 due to flash size limitations.

ArtCodeZen, 2024-07-30:
Ok, thank you for information!

closed 2024-07-30 by ArtCodeZen