speeduino/speeduino #1235: IAC Running Hot on Firmware 202305 and newer

dMartin1954, 2024-08-13:
I recently restarted a project I had been working on and discovered that using any firmware version starting with 202305 or newer that my IAC Stepper would run hot even though it was set “When Active” for stepper power.  What I discovered is that the Enable pin on the stepper driver was not being set high when the stepping action was completed causing voltage to be applied to the coils continuously.  On versions 202305 and 202310, if I set the minimum number of steps to one, the Enable pin is set high when stepping is completed, using a number greater then one would cause the Enable to not be set high when the stepping is completed.  On version 202402.1 even with minimum steps set to one,  the Enable pin would not be set high when the stepping was completed.  I am running the idle in open loop mode and varying the coolant temperature using a simulator to test the IAC stepper operation.  I am monitoring the stepper steps using the IAC Load gauge in Tuner Studio and using a voltmeter to monitor the state of the Enable pin on the driver.  The stepper is a GM 4 wire stepper which is operating correctly other than getting hot.

I looked at the idle code and found there was a condition check to see if the stepper is at the final step in the cycle.  I suspect that the condition is not being met when missed steps are set to anything greater than one.  I removed this final step check so that the only condition to set the Enable high is that “When Active” is set and this seems to have fixed the issue in my application, but I am not sure if this has any other affect on the stepper action since I am a novice when it comes to C++ programming.  It did appear that in the code anytime a step was initiated, the Enable pin was explicitly set low.

noisymime, 2024-08-13:
Suspect this is the same as #1203 

Are you running open loop stepper control?

dMartin1954, 2024-08-13:
Yes, open loop.  This does appear to be the same a #1203.

commit ab3eca07 (2024-08-13): Fix stepper driver on open loop always remaining enabled

Also fixes #1235
commit bca98cd2 (2024-08-13): Fix stepper driver on open loop always remaining enabled

Also fixes #1235
closed 2024-08-13 by noisymime