speeduino/speeduino #1240: sequential fuel does not work in rover decoder

dwisnie2, 2024-08-16:
version 202402.1.

Injection pattern is sequential but there are still 2 squirts per engine cycle (like the engine spins twice as fast). my cam sensor and config is right - i can see it on tooth logger. Bug confirmed on car and ardustim. I dont know if sequential ignition is working right.

[image]

RempageR1, 2024-09-02:
Does it say fully sequential in tuner studio when the engine runs?

Nikolai521, 2024-09-06:
You can try setting it to 2 squirts per cycle?
In the manual states that Speedy divides it by 1 internally
Im running on sequential and 2 squirts and im getting the 4 squirts

dwisnie2, 2024-10-02:
> Does it say fully sequential in tuner studio when the engine runs?
I am not sure where i can find that indicator, i got full sync even when i unplug the cam sensor.

> You can try setting it to 2 squirts per cycle? In the manual states that Speedy divides it by 1 internally Im running on sequential and 2 squirts and im getting the 4 squirts (1 per injector)

I was trying 2 but no effect, by 2 squirts i mean every injector squirts 2 times per engine cycle. Is this the way this setting work ?

I have my own board computer reading RPM based on injector signal, and it shows the same value when i set semi-sequental and full sequential

RempageR1, 2024-10-03:
[image]
When you have half sync, the status indicator above should be yellow. Does it also say full sync if you start with the cam sensor off?

dwisnie2, 2024-10-03:
> ![afbeelding](https://private-user-images.githubusercontent.com/103254506/373355583-4a4e3071-a0dd-40cd-bc49-b00a81e62d3e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjc5Nzc2NTAsIm5iZiI6MTcyNzk3NzM1MCwicGF0aCI6Ii8xMDMyNTQ1MDYvMzczMzU1NTgzLTRhNGUzMDcxLWEwZGQtNDBjZC1iYzQ5LWIwMGE4MWU2MmQzZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAwM1QxNzQyMzBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02MmU5YmVkYzYxNTM3MTJkYjU2ZjE5MTZjOTEwZGY3OGNhNDZhMWFiZTg3OGQ0ZTliNDNmNTExZjMxODczY2JkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.3v8uv82ua56Hmpg_uMMJHEOL6VjirZGcZUS7q9sbKlA) When you have half sync, the status indicator above should be yellow. Does it also say full sync if you start with the cam sensor off?

Yes. I never seen it showing half sync. btw i do not have 14-14 pattern - wasted spark works ok.

mike501, 2024-10-04:
Hello, I wrote the original rover code as my first ever decoder. On its submission into the codebase it did do sequential fuel and spark. I used to run an logic monitor on the trigger pins of the arduino so I could see it fired the relevant pins correctly.

If you're removing the cam sensor to check for half sync, ensure you ground the connection. Any noise could be taken as a cam signal, if it arrived at the right time the system would find sync & then its off running sequential. The code can be too clever for its own good (i got carried away writing it).

Its been a while since I've looked at speeduino but i'll try to dig a board out, connect up the logic monitor and see what its doing with the latest firmware.

What car is it in & how have you connected Speeduino to the engine loom? If you've found a source of ECU connectors please contact me privately.

mike501, 2024-10-05:
Done a quick set of tests - definitely a problem with it allowing sequential when it shouldn't (ie without a secondary cam signal).

However using this to my advantage, with just a crank trigger i can select sequential fuel and get over 720 degrees 4 injection events (hopefully the image attaches). I think this is correct, please let me know if its not.
[image]

This is with squirts per engine cycle set to 2 (which if i recall correctly is ignored in sequential so it just does as shown 1 injection squirt per cycle per cylinder).

dwisnie2, 2024-10-05:
It looks great on your waveforms. I will investigate it once again with scope and post the results here. I am sure i was getting good cam signal, it was visible in trigger log

mike501, 2024-10-05:
fix to sequential here https://github.com/mike501/speeduino/tree/RoverSequentialFix let me know how you get on.

dwisnie2, 2024-10-06:
> fix to sequential here https://github.com/mike501/speeduino/tree/RoverSequentialFix let me know how you get on.

turns out it was all good from the begining. I make mistake somewhere about this frequency, here is waveform of sequential working ok on real engine

[image]

I confirmed that on official newest relase there is no way to get half sync, speeduino is making sequential with cam sensor unpluged. 
On yours fixed decoder starting without cam sensor gives halfsync and rollback to semisequental, after plugging it in it goes back to sequential + full sync. unpluging while engine is idleing does not make any change (i think that is correct)  

semiseq waveform:
[image]

To this day i was running semi-sequental with number of squirts set to 1 🤡I don't know how but the engine somehow runs ok

mike501, 2024-10-06:
Once the engine has full sequential it shouldn't need the cam any more as its aligned itself to the correct phasing of the engine.

Can you get some miles on the new decoder so we can be sure its fine & then i'll create a request to the master updated.

dwisnie2, 2024-10-08:
I drove about 200km with this firmware, and everything seems to be fine. It should be safe to merge this fix to master.
Thank you.

commit 406e8806 (2024-10-18): Fix sequential sync detection - raised in issue 1240 (#1256)

Fixes #1240
closed 2024-10-18 by noisymime