speeduino/speeduino #1243: Miata decoder - sequential does not run properly above 5000rpm

AlexE75, 2024-09-02:
Hello,

I have noticed issue on Mazda MX-5 Miata. Tested on 3 different cars, all 2001-2005 mk2.5. Above around 5000rpm there is massive unstability. Logs show RPM random spikes above 10000rpm or so

Three dyno runs of the same car, one by one, without changing anything
[image]
[mx51.8vvt problem.pdf](https://github.com/user-attachments/files/16838464/mx51.8vvt.problem.pdf)

logs and tune attached

this car is running perfectly fine on stock ecu
[mx5 1.8vvt stock.pdf](https://github.com/user-attachments/files/16838457/mx5.1.8vvt.stock.pdf)

Identical issue on 3 cars I had access to. All running ok on stock ECU

And all 3 cars can run ok on Speeduino - after switching from sequential to semi-sequential. But this makes problem if using larger injectors.

kind regards
Aleksander

[mx5 mk25 18vvt issue.zip](https://github.com/user-attachments/files/16838441/mx5.mk25.18vvt.issue.zip)

noisymime, 2024-09-03:
Have you tried different values on the crank filter? At the moment you've got it set to off, so it's relying on the hardware filter only. It's very odd that this only happen with sequential though.

Was it the same ECU you used on all 3 cars? 

I haven't immediately been able to reproduce this on the bench, but will have more of a play switching between semi-sequential and sequential to see what difference I can find.

AlexE75, 2024-09-04:
Hello,

yes, I tried different crank filter settings, starting from medium. And other options like "re-sync every cycle". Didnt helped.

I used 2 ECUS, at first it was version1, then I tried version2 (different output drivers and thus redesigned trace layout). Behaviour was the same.

Tested 202310 and 202402 firmware, the same effect

noisymime, 2024-10-09:
I've had a good play with this on the bench and cannot reproduce your issue. I'm seeing smooth fuel and ignition delivery on sequential with the Miata 99-05 trigger all the way up to 10,000rpm (Haven't tested beyond there). 

One thing I have noticed in your tune though is that you have the Flat Shift set for 5000 (Which is the point you're saying you get problems from) and critically the pullup resistor setting set to 'Float'. Having that as float means that you MUST have an external pullup resistor on the board that you're using otherwise the flat shift cutout will come on basically at random above the 5000rpm point, which could be what you're seeing. 
The majority of setups should have this set to 'Pullup' rather than float, but it depends exactly on what hardware you're using.

Can you try this again with the Flat Shift either disabled or the pullup set correctly (Assuming you don't have an external pullup resistor).

noisymime, 2024-10-09:
Just around the noise spikes that you're seeing, what board designs are you running with and, if you know them,  what are the values of the crank/cam filter caps?

AlexE75, 2024-10-10:
Hi,

I have pullup resistor (4k7 to +5V) on clutch input for sure. If clutch would be the case switching to semi-sequential would not help, am I right?

values for CKP and CMP filtering on my boards:

CKP
pullup 1k, series 470r, parallel 10n

CMP
pullup 1k, series 470r, parallel 22n

kind regards
Aleksander