speeduino/speeduino #1245: Mega2560 6x3 Sequential INJ + Waste COP MODE - somewhat broken

ggurov, 2024-09-12:
The setup is 6x3 Mega2560 where the waste spark is accomplished via individual coil on plug, connected to separate pins.

specifically for 6 cylinder engine with 6 pins defined. the following needs to be added for it to work as expected:
```

      pinCoil1 = 40; //Pin for coil 1
      pinCoil2 = 38; //Pin for coil 2
      pinCoil3 = 52; //Pin for coil 3
      pinCoil4 = 48; //Pin for coil 4
      pinCoil5 = 36; //Pin for coil 5 PLACEHOLDER value for now
      pinCoil6 = 34;
      pinMode(pinCoil5, OUTPUT);
      pinMode(pinCoil6, OUTPUT);
```
specifically the pinMode() calls to set pin to OUTPUT. there's a check down in init.cpp that will do the same if IGN_CHANNELS >=5 or >= 6 which does not happen with 6x3, as that sets IGN_CHANNELS=3 

after pinMode() is called signals are output correctly.

noisymime, 2024-09-19:
As mentioned in #1246, Wasted COP is only intended to work where there are sufficient ignition channels natively on the board. 

**Eg** For Wasted COP on a 4 cylinder you need 4 ign channels. A 6 cylinder needs 6 channels etc. 

So Wasted COP on a 6 cylinder using the 6+3 mode is not currently supported. I will track this as a feature request to look at supporting this case though.

pazi88, 2024-09-20:
This has worked fine for very long time earlier. At least on 202305 it works, and there is hundreds of boards out there that rely on this feature (the bmw-pnp). Boards have 6-6 channels, because with stm32 you can do full sequential. But it works also with arduino mega in wasted cop mode, because the timer channel requirement is only 6+3 on wasted cop.